ARISE-Initiative/robosuite #198: gym.wrappers.Monitor video recording compatibility

bebbo203, 2021-03-01:
Since it's not possible to record a video of the environments using the gym wrappers, this could be a solution:

After creating a robosuite environment with `has_offscreen_render = True`:

`env = GymWrapper(e)
env.metadata = {'render.modes': ['human', 'rgb_array', 'depth_array'], 'video.frames_per_second': 67}`

After that, just change in the file robosuite/envirnoments/base.py the function render() to this:

`def render(self,
               mode='human',
               width=500,
               height=500,
               camera_id=0,
               camera_name=None):

        if mode == 'rgb_array':
            #self.viewer = MjRenderContextOffscreen(self.sim, -1)
            
            self.sim._render_context_offscreen.render(width, height, camera_id=camera_id)
            data = self.sim._render_context_offscreen.read_pixels(width, height, depth=False)
            # original image is upside-down, so flip it
            return data[::-1, :, :]
       
        elif mode == 'human':
            self.viewer.render()`

Is this an acceptable solution?

cremebrule, 2021-03-04:
Hi @bebbo203 ,

I don't see why this shouldn't work. However, a couple caveats:

1. `human` mode won't work unless you have `has_renderer = True`
2. We haven't tested the exact method you're suggesting for rendering / reading pixels, so we can't guarantee that this is a stable method. We've unfortunately had a bunch of issues with the MjRenderer in the past ("black screen of death"), so we try to stick to what we know works best. Is there a reason why you can't directly render using `sim.render(cam_name, width, height, depth)`?

bebbo203, 2021-03-05:
I can't use that since I needed the Gym monitor wrapper to record the environment

cremebrule, 2021-03-06:
What about using another minimal wrapper to copy the image obs over into the `info` section that you can then parse as you need? e.g.:

```python
import robosuite as suite
from robosuite.wrappers import Wrapper, GymWrapper

class ImageInfoWrapper(Wrapper):
    def step(self, action):
        ob_dict, reward, done, info = self.env.step(action)
        info["img_frames"] = obs_dict["image-state"]
        return ob_dict, reward, done, info

# Create env, wrap with image info wrapper, then with gym wrapper
env = GymWrapper(ImageInfoWrapper(suite.make(...)))
```

yukezhu, 2021-04-27:
Closing this as no follow-up in a month.

closed 2021-04-27 by yukezhu