ARISE-Initiative/robosuite #199: Can I use camera obervation with 1 GPU

huang1997214, 2021-03-03:
When I make the environment.

env = nl_guide_lift_env(
    robots="Sawyer",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=False,
    has_offscreen_renderer=True,
    use_camera_obs=True,
    ignore_done=True,
    render_visual_mesh=True,
    control_freq=20,
)

The following error occurs
  File "mujoco_py/mjrendercontext.pyx", line 45, in mujoco_py.cymj.MjRenderContext.__init__
  File "mujoco_py/mjrendercontext.pyx", line 113, in mujoco_py.cymj.MjRenderContext._setup_opengl_context
  File "mujoco_py/opengl_context.pyx", line 130, in mujoco_py.cymj.OffscreenOpenGLContext.__init__
RuntimeError: Failed to initialize OpenGL

This seems to result in 1 GPU. I wondering if there is some way that I can use 1GPU to get the image observation.
I want to use image observation to train a DQN.
Thanks!

hermanjakobsen, 2021-03-04:
This is a known issue with mujoco-py and rendering. I have posted a similar [issue](https://github.com/ARISE-Initiative/robosuite/issues/114) some time ago :)

huang1997214, 2021-03-04:
Sorry, I am not familiar with mujoco-py.

closed 2021-03-04 by huang1997214

RayYoh, 2021-11-24:
> This is a known issue with mujoco-py and rendering. I have posted a similar [issue](https://github.com/ARISE-Initiative/robosuite/issues/114) some time ago :)

So how to solve this problem?

peterdavidfagan, 2021-11-25:
@RayYoh it is a bit of a hack but you can use the following from my experience: https://github.com/openai/mujoco-py/issues/268#issuecomment-968205448

Kudo to @HesNobi

RayYoh, 2021-11-26:
> @RayYoh it is a bit of a hack but you can use the following from my experience: [openai/mujoco-py#268 (comment)](https://github.com/openai/mujoco-py/issues/268#issuecomment-968205448)
> 
> Kudos to @HesNobi for this.

Thanks a lot. Hope it works.