ARISE-Initiative/robosuite #201: What's the unit of the robot joints?

guodashun, 2021-03-18:
Hello, today I try to control the velocities of robot joints with `controller_config = load_controller_config(default_controller='JOINT_VELOCITY')`, and I find the default value of the maximum velocity is 1. But I cannot find the specific unit of the velocity. So the question is what's the unit of the robot joints, and how can I control the joint with a specific velocity (for example, 3m/s)?

yukezhu, 2021-03-18:
The action space is normalized to [-1, 1] for convenience of policy learning. In practice, the real numerical range is defined in controller config files, e.g., this one for joint velocity: https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/config/joint_velocity.json

Please check the MuJoCo documentations to see the units of the output.

guodashun, 2021-03-19:
Thanks for your response. But I just met another confusing case.
I set my vel_control.json as follows:
```json
{
    "type" : "JOINT_VELOCITY",
    "input_max": 1,
    "input_min": -1,
    "output_max": 1,
    "output_min": -1,
    "kp": 3,
    "velocity_limits": [-1,1],
    "interpolation": null,
    "ramp_ratio": 0.2
}
```
and I gave an action `[1,0,0,0,0,0,0]`. Then I got the velocity of the joint from the observation. With the iteration by `env.step(action)`, the velocity of the first joint was increasing and exceeded 1.
Then I changed my json file as follows:
```json
{
    "type" : "JOINT_VELOCITY",
    "input_max": 10,
    "input_min": -10,
    "output_max": 10,
    "output_min": -10,
    "kp": 3,
    "velocity_limits": [-1,1],
    "interpolation": null,
    "ramp_ratio": 0.2
}
```
The velocity I got was steady at 1 and never exceeded 1 again.
What's the problem?

cremebrule, 2021-04-03:
Hi @guodashun ,

Thanks for the additional info. There may perhaps be a miscommunication in terms of what `velocity_limits` does. In the `.json` file you specified, the `velocity_limits` clamps the commanded velocity targets; however, this has no bearing on what observations (i.e.: "real" velocities) are received during the simulation.

That being said, in either case the observed behavior shouldn't differ between either of your configs. When you were using the first config (with input / outputs set to +/- 1), did the velocity converge to 1 or did it continue to exceed past 1 indefinitely?

guodashun, 2021-04-03:
I got the first joint's velocity by
```python
action = [1, 0,0,0,0,0]
obs, _,_,_ = env.step(action)
print("first joint vel: ",obs['robot0_joint_vel'][0])
```
With my first config which _in/output max/min_ is 1, I got the following results:
```
first joint vel:  0.5438005821382699
first joint vel:  0.9773168302500089
first joint vel:  1.0119495290541698
first joint vel:  1.0318542003072435
first joint vel:  1.0517587183525272
first joint vel:  1.0716623228610849
first joint vel:  1.0915649755693975
first joint vel:  1.1114666572300547
first joint vel:  1.1313673501402979
first joint vel:  1.1512670367663533
first joint vel:  1.1711656996301343
first joint vel:  1.1910633212981812
first joint vel:  1.210959884379
first joint vel:  1.2308553715211679
first joint vel:  1.250749765411594
first joint vel:  1.2706430487738651
first joint vel:  1.2905352043666805
first joint vel:  1.3104262149823651
......
first joint vel:  3.5466141346477182
first joint vel:  3.566265912563609
first joint vel:  3.585914661205437
first joint vel:  3.6055603647500245
first joint vel:  3.6252030073841093
first joint vel:  3.6448425733043934
first joint vel:  3.6644790467176005
first joint vel:  3.684112411840527
first joint vel:  3.703742652900095
first joint vel:  3.7233697541334054
first joint vel:  3.7429936997877866
first joint vel:  3.762614474120851
first joint vel:  3.0056911711757035
first joint vel:  -0.15625702866135063
first joint vel:  -0.41459127560476966
first joint vel:  -0.2706233968259257
```
At last the velocity became to negative because of the limit position.
Then I changed the config to the another one which _in/output max/min_ is 10, and the result is:
```
first joint vel:  0.54386789012069
first joint vel:  0.9773452743042691
first joint vel:  1.0054636894187108
first joint vel:  1.0057679685121814
first joint vel:  1.005495318014982
first joint vel:  1.0052241272952527
first joint vel:  1.0049660738365804
first joint vel:  1.0047207618439886
first joint vel:  1.0044875675406397
first joint vel:  1.0042658924792671
first joint vel:  1.0040551676406648
first joint vel:  1.003854852109625
first joint vel:  1.0036644316907015
first joint vel:  1.0034834175885048
first joint vel:  1.003311345153008
first joint vel:  1.0031477726868248
first joint vel:  1.0029922803113973
first joint vel:  1.0028444688891993
first joint vel:  1.0027039589991746
first joint vel:  1.0025703899627878
first joint vel:  1.0024434189181888
first joint vel:  1.0023227199401068
first joint vel:  1.0022079832032242
first joint vel:  1.0020989141868737
first joint vel:  1.0019952329190251
first joint vel:  1.0018966732576156
first joint vel:  1.0018029822073802
......
```
The velocity seemd to converge to 1 finally.

cremebrule, 2021-04-04:
Can you provide a reproducible script? I'm testing now with Panda + Lift env and I can't replicate the issue using your controller configs.

guodashun, 2021-04-04:
I uploaded the [test code](https://github.com/guodashun/ur5-planning) in my repository.
You can run this test demo by
```bash
$ python test.py
```
Thanks!

PR #236 (2021-09-22): Fix Joint Velocity Bug
Fixes #201. Removes the joint velocity observation clipping during the controller computation causing inaccurate results, and increases the default kp gain value for faster convergence.
closed 2021-09-22 by yukezhu