ARISE-Initiative/robosuite #202: can you share the code in VICES?

csufangyu, 2021-03-23:
hi I read the paper [Variable Impedance Control in End-Effector Space: An Action Space for Reinforcement Learning in Contact-Rich Tasks](https://arxiv.org/abs/1906.08880), nut I not find the code in paper,the code seems to only open source the training environment.
thanks very much!

yukezhu, 2021-03-24:
Please check out this branch: https://github.com/ARISE-Initiative/robosuite/tree/vices_iros19

closed 2021-03-24 by yukezhu

csufangyu, 2021-03-25:
yes I'm reading this branch,but in the code, I can not find the train file,and i find the [config file](https://github.com/ARISE-Initiative/robosuite/tree/vices_iros19/robosuite/scripts/config),but not find how to train or use it. can you help me?thanks very much.