ARISE-Initiative/robosuite #207: Ram usage keeps increasing when training with ray rllib

ava6969, 2021-04-02:
when i train with ray rllib, The ram usage keep increasing, is there something that needs to be cleaned up. ray works in other environments easily.

cremebrule, 2021-04-03:
Hi @ava6969 ,

Can you provide more info? What robosuite task are you running? What rllib algo are you using? A reproducible script would be greatly appreciated.

ava6969, 2021-04-03:
> Hi @ava6969 ,
> 
> Can you provide more info? What robosuite task are you running? What rllib algo are you using? A reproducible script would be greatly appreciated.

I am doing a simple task with stacking.
i am using  ray tune and rllib with PPO , same thing happens with other algorithms. I am using multiple workers 30 + .
do you think it may be from ray rllib

cremebrule, 2021-04-04:
Hmm, I'm not entirely sure. I've used rllib in the past and I don't recall running into this issue. Can you check if you get the same issue using an older version of robosuite (v1.0 or v1.1)?

ava6969, 2021-04-10:
I reverted back. i still get the same error with previous versions.

"""
This file implements a wrapper for facilitating compatibility with OpenAI gym.
This is useful when using these environments with code that assumes a gym-like
interface.
"""
import time
from collections import OrderedDict

import numpy as np
from gym import spaces
from robosuite.wrappers import Wrapper

class BasicMultiDiscrete(Wrapper):
    env = None

    def __init__(self, env, keys=None, action_bin=11):
        """
        Initializes the Gym wrapper.

        Args:
            env (MujocoEnv instance): The environment to wrap.
            keys (list of strings): If provided, each observation will
                consist of concatenated keys from the wrapped environment's
                observation dictionary. Defaults to robot-state and object-state.
        """
        super().__init__(env)
        self.env = env

        obs = env.reset()
        # set up observation and action spaces
        excludes = {'robot0_robot-state', 'object-state'}

        shapes = {k: spaces.Box(low=-np.inf, high=np.inf, shape=obs[k].shape, dtype=np.float32)
                  for k in obs.keys() if k not in excludes}

        self.observation_space = spaces.Dict(shapes)
        low, high = self.env.action_spec

        self.actions_map = np.array([np.linspace(-1, 1, action_bin) for _ in range(len(low))])
        self.action_space = spaces.MultiDiscrete([action_bin] * len(low))

    def format(self, obs_dict:OrderedDict):
        """
        Filters keys of interest out and concatenate the information.

        Args:
            obs_dict: ordered dictionary of observations
        """

        obs_dict.pop('robot0_robot-state')
        obs_dict.pop('object-state')
        return dict(obs_dict)

    def reset(self):
        ob_dict = self.env.reset()
        return self.format(ob_dict)

    def step(self, action):
        actions = np.take(self.actions_map, action)
        ob_dict, reward, done, info = self.env.step(actions)
        return self.format(ob_dict), reward, done, info

from ray.rllib import SampleBatch
from ray.rllib.models.torch.torch_modelv2 import TorchModelV2
from ray.rllib.models import ModelCatalog, ModelV2
from ray.rllib.models.torch.misc import SlimFC, normc_initializer
from ray.rllib.policy.view_requirement import ViewRequirement
from ray.rllib.utils import override
[... 110 lines cut ...]
    config["rollout_fragment_length"] = 10
    config["entropy_coeff"] = 0.01
    config["lambda"] = 0.95
    config["sgd_minibatch_size"] = 32000
    config["num_sgd_iter"] = 60
    # config["remote_worker_envs"] = True
    config["clip_param"]  = 0.2

    tune.run(
        'PPO',
        name=model_name,
        local_dir='results',
        config=config,
        reuse_actors=True,
        checkpoint_at_end=True,
        verbose=3,
        # restore=RESTORE_PATH,
        # restore='/home/dewe/PycharmProjects/robotics_drl/results/robot_20210329-18h17/PPO_multidiscrete_robot-v0_9d8b0_00000_0_2021-03-29_18-17-57/checkpoint_920/checkpoint-920',
        checkpoint_freq=10
    )

    ray.shutdown()

if __name__ == '__main__':
    main()

ava6969, 2021-05-04:
i think its a ray problem dude to to their system, i used a custom implementation

closed 2021-05-04 by ava6969