ARISE-Initiative/robosuite #214: Orientation limit in OSC

Wenxuan-Zhou, 2021-04-15:
Is it possible to provide an example or some explanation of the "orientation limits" in [OSC](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/controllers/osc.py)? I tried to set the orientation limit and ran demo_device_control.py, but it has very strange behavior.
In [this line](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/utils/control_utils.py#L246), why is it [euler[1], euler[0], euler[2]] instead of [euler[0], euler[1], euler[2]]?

Thanks in advance!

yukezhu, 2021-04-27:
Thanks for opening this issue. We are investigating this problem and will get back to you soon.

yukezhu, 2021-05-13:
This has been fixed. Thank you for reporting the issue!