ARISE-Initiative/robosuite #217: Rendering deformable objects with skin attribute will raise segmentation fault

JieFeng-cse, 2021-04-25:
There is a problem with the official mujoco_py that when you are rendering a deformable object with a skin attribute, it will raise segmentation fault. Many people have met this problem and it is solved.
https://github.com/openai/mujoco-py/issues/373
I think the core team should be aware of that. To make a satisfying environment for manipulating deformable objects, some bugs of mujoco_py have to be solved.

yukezhu, 2021-04-25:
Thanks for the heads-up. Is this still causing the segfault issue with the latest mujoco-py version?

JieFeng-cse, 2021-04-26:
Yes, I have tested mujoco-py 2.0.2.13, and I met the segment fault again, so I raise the issue.  However, there is also no need to worry. It is a simple problem just need to be aware of. 
Use codes from Herman can easily solve the problem. https://github.com/hermanjakobsen/mujoco-py
`pip3 install --user --force-reinstall --no-compile git+https://github.com/hermanjakobsen/mujoco-py` will do the trick. So it's okay to update to the latest mujoco-py. When I have time, I will try to find out how they solved the bug. ( It should be just a few lines)

yukezhu, 2021-04-27:
Thanks for the information. I am closing this issue for now. If you find how they fixed this bug, please feel free to re-open or create an issue to report this.

closed 2021-04-27 by yukezhu