ARISE-Initiative/robosuite #218: In OSC_POSE control, there is coupling of XYZ axis motion.

kaixindelele, 2021-04-27:
In osc_pose control, there is coupling of XYZ axis motion.
Test experiments are as follows.
1. static experiment, i.e. action = np.zeros(action_dim)
All the axes at the end remain stable and do not jitter, the system is stable.
2. Give only a value of -0.05 to the x-axis, the robot arm will move in the negative direction of the x-axis, but the y and z-axes will also have a small offset.
3. After 100 steps, the eef delta moving is [-0.06962994  0.00422471  0.0119571 ].

This phenomenon I also recorded the screen before：[https://github.com/ARISE-Initiative/robosuite/issues/124](https://github.com/ARISE-Initiative/robosuite/issues/124)

Is there any way to try to mitigate this phenomenon?

the demo code:
```
from robosuite.utils.input_utils import *

if __name__ == "__main__":
    options = dict()
    # Choose environment and add it to options
    envs = sorted(suite.ALL_ENVIRONMENTS)

    for e in range(12, 13):
        print("task_name:", envs[e])
        options["env_name"] = envs[e]
        options["robots"] = 'Sawyer'
        # Hacky way to grab joint dimension for now
        joint_dim = 6 if options["robots"] == "UR5e" else 7
        # Choose controller
        controller_name = 'OSC_POSE'
        # Load the desired controller
        options["controller_configs"] = suite.load_controller_config(default_controller=controller_name)
        # Define the pre-defined controller actions to use (action_dim, num_test_steps, test_value)
        controller_settings = {
            "OSC_POSE":         [6, 6, 0.1],
            "OSC_POSITION":     [3, 3, 0.1],
            "IK_POSE":          [6, 6, 0.01],
            "JOINT_POSITION":   [joint_dim, joint_dim, 0.2],
            "JOINT_VELOCITY":   [joint_dim, joint_dim, -0.1],
            "JOINT_TORQUE":     [joint_dim, joint_dim, 0.25]
        }

        # Define variables for each controller test
        action_dim = controller_settings[controller_name][0]
        num_test_steps = controller_settings[controller_name][1]
        test_value = controller_settings[controller_name][2]

        # Define the number of timesteps to use per controller action as well as timesteps in between actions
        steps_per_action = 15
        steps_per_rest = 15
        print("Press \"H\" to show the viewer control panel.")
        try:
            env = suite.make(
                    **options,
                    has_renderer=True,
                    has_offscreen_renderer=False,
                    ignore_done=True,
                    use_camera_obs=False,
                    horizon=(steps_per_action + steps_per_rest) * num_test_steps,
                    control_freq=10,
                    gripper_types="RethinkGripper",
                )

            for i_episode in range(1):
                observation = env.reset()
                for t in range(100):
                    if t == 0:
                        origin_eef = observation['robot0_eef_pos']
                    env.render()
                    action = np.zeros(action_dim+1)
                    action[0] = -0.05
                    print("t:", t, end='\t')
                    observation, reward, done, info = env.step(action)
                    print("eef:", np.round(observation['robot0_eef_pos'], 4))
                    if done:
                        print("Episode finished after {} timesteps".format(t + 1))
                        break
                print("delta_moving:", observation['robot0_eef_pos'] - origin_eef)

            # Shut down this env before starting the next test
            env.close()
        except Exception as e:
            print("env_error:", e)
            env.close()
        env.close()

```

print log:
```
t: 0	eef: [-0.1412  0.158   1.0057]
t: 1	eef: [-0.1419  0.158   1.0057]
t: 2	eef: [-0.1426  0.158   1.0058]
t: 3	eef: [-0.1433  0.1581  1.006 ]
t: 4	eef: [-0.144   0.1581  1.0061]
t: 5	eef: [-0.1447  0.1581  1.0062]
t: 6	eef: [-0.1454  0.1581  1.0064]
t: 7	eef: [-0.146   0.1582  1.0065]
t: 8	eef: [-0.1467  0.1582  1.0066]
t: 9	eef: [-0.1474  0.1582  1.0067]
t: 10	eef: [-0.1481  0.1583  1.0069]
t: 11	eef: [-0.1488  0.1583  1.007 ]
t: 12	eef: [-0.1495  0.1583  1.0071]
t: 13	eef: [-0.1502  0.1583  1.0072]
t: 14	eef: [-0.1509  0.1584  1.0074]
t: 15	eef: [-0.1516  0.1584  1.0075]
t: 16	eef: [-0.1523  0.1584  1.0076]
t: 17	eef: [-0.153   0.1584  1.0077]
t: 18	eef: [-0.1537  0.1585  1.0079]
t: 19	eef: [-0.1544  0.1585  1.008 ]
t: 20	eef: [-0.1551  0.1585  1.0081]
t: 21	eef: [-0.1558  0.1585  1.0082]
t: 22	eef: [-0.1565  0.1586  1.0083]
t: 23	eef: [-0.1572  0.1586  1.0085]
t: 24	eef: [-0.1579  0.1586  1.0086]
t: 25	eef: [-0.1586  0.1587  1.0087]
t: 26	eef: [-0.1593  0.1587  1.0088]
t: 27	eef: [-0.16    0.1587  1.009 ]
t: 28	eef: [-0.1607  0.1588  1.0091]
t: 29	eef: [-0.1614  0.1588  1.0092]
t: 30	eef: [-0.1621  0.1589  1.0093]
t: 31	eef: [-0.1628  0.1589  1.0094]
t: 32	eef: [-0.1635  0.1589  1.0096]
t: 33	eef: [-0.1642  0.159   1.0097]
t: 34	eef: [-0.1649  0.159   1.0098]
t: 35	eef: [-0.1656  0.1591  1.0099]
t: 36	eef: [-0.1663  0.1591  1.01  ]
t: 37	eef: [-0.167   0.1591  1.0102]
t: 38	eef: [-0.1677  0.1592  1.0103]
t: 39	eef: [-0.1684  0.1592  1.0104]
t: 40	eef: [-0.1691  0.1593  1.0105]
t: 41	eef: [-0.1698  0.1593  1.0106]
t: 42	eef: [-0.1705  0.1594  1.0108]
t: 43	eef: [-0.1712  0.1594  1.0109]
t: 44	eef: [-0.1719  0.1595  1.011 ]
t: 45	eef: [-0.1726  0.1595  1.0111]
t: 46	eef: [-0.1733  0.1596  1.0113]
t: 47	eef: [-0.174   0.1596  1.0114]
t: 48	eef: [-0.1747  0.1597  1.0115]
t: 49	eef: [-0.1754  0.1597  1.0116]
t: 50	eef: [-0.1761  0.1598  1.0117]
t: 51	eef: [-0.1768  0.1598  1.0119]
t: 52	eef: [-0.1775  0.1599  1.012 ]
t: 53	eef: [-0.1782  0.1599  1.0121]
t: 54	eef: [-0.1789  0.16    1.0122]
t: 55	eef: [-0.1796  0.16    1.0123]
t: 56	eef: [-0.1803  0.1601  1.0125]
t: 57	eef: [-0.181   0.1601  1.0126]
t: 58	eef: [-0.1817  0.1601  1.0127]
t: 59	eef: [-0.1824  0.1602  1.0128]
t: 60	eef: [-0.1831  0.1602  1.0129]
t: 61	eef: [-0.1838  0.1603  1.0131]
t: 62	eef: [-0.1845  0.1603  1.0132]
t: 63	eef: [-0.1852  0.1604  1.0133]
t: 64	eef: [-0.1859  0.1604  1.0134]
t: 65	eef: [-0.1866  0.1605  1.0135]
t: 66	eef: [-0.1873  0.1605  1.0137]
t: 67	eef: [-0.188   0.1606  1.0138]
t: 68	eef: [-0.1887  0.1606  1.0139]
t: 69	eef: [-0.1894  0.1607  1.014 ]
t: 70	eef: [-0.1901  0.1607  1.0141]
t: 71	eef: [-0.1908  0.1608  1.0142]
t: 72	eef: [-0.1915  0.1608  1.0144]
t: 73	eef: [-0.1922  0.1609  1.0145]
t: 74	eef: [-0.1929  0.1609  1.0146]
t: 75	eef: [-0.1936  0.161   1.0147]
t: 76	eef: [-0.1943  0.161   1.0148]
t: 77	eef: [-0.195   0.1611  1.015 ]
t: 78	eef: [-0.1957  0.1611  1.0151]
t: 79	eef: [-0.1963  0.1612  1.0152]
t: 80	eef: [-0.197   0.1612  1.0153]
t: 81	eef: [-0.1977  0.1613  1.0154]
t: 82	eef: [-0.1984  0.1613  1.0156]
t: 83	eef: [-0.1991  0.1614  1.0157]
t: 84	eef: [-0.1998  0.1614  1.0158]
t: 85	eef: [-0.2005  0.1615  1.0159]
t: 86	eef: [-0.2012  0.1615  1.016 ]
t: 87	eef: [-0.2019  0.1616  1.0161]
t: 88	eef: [-0.2026  0.1616  1.0163]
t: 89	eef: [-0.2033  0.1617  1.0164]
t: 90	eef: [-0.204   0.1617  1.0165]
t: 91	eef: [-0.2047  0.1618  1.0166]
t: 92	eef: [-0.2054  0.1618  1.0167]
t: 93	eef: [-0.2061  0.1619  1.0169]
t: 94	eef: [-0.2068  0.1619  1.017 ]
t: 95	eef: [-0.2075  0.162   1.0171]
t: 96	eef: [-0.2082  0.162   1.0172]
t: 97	eef: [-0.2089  0.1621  1.0173]
t: 98	eef: [-0.2096  0.1621  1.0174]
t: 99	eef: [-0.2103  0.1622  1.0176]
delta_moving: [-0.06962994  0.00422471  0.0119571 ]

```

roberto-martinmartin, 2021-05-13:
Hi! You could try increasing the proportional gains of the controller and adapting the differential gain to remain critically damped. Let me know if the error decreases

kaixindelele, 2021-05-17:
> Hi! You could try increasing the proportional gains of the controller and adapting the differential gain to remain critically damped. Let me know if the error decreases

Hi, thanks for your reply, I tried to adjust the kp parameter and found that the error was not reduced too much and there was no obvious pattern.
Then I saw that there is another parameter uncouple_pos_ori in the osc_pose.json file, and when I set this parameter to false, there is an order of magnitude reduction in the pos-xyz axis coupling.

"""
kp: 90
delta_moving: [-0.04402996  0.00353451  0.01451138]
kp: 120
delta_moving: [-0.05587618  0.002599    0.01547164]
kp: 150
delta_moving: [-0.07227131  0.00324817  0.01523023]
kp: 180
delta_moving: [-0.08256543  0.00534876  0.01342693]
kp: 210
delta_moving: [-0.09550479  0.00489477  0.0129235 ]
kp: 240
delta_moving: [-0.10363343  0.00497017  0.01187464]
delta_moving: [-0.10263296  0.005566    0.0116839 ]
kp: 270
delta_moving: [-0.11214408  0.00367714  0.01034413]

kp: 270---"uncouple_pos_ori": false,
delta_moving: [-0.14572088  0.00026412 -0.00040424]
kp: 150---"uncouple_pos_ori": false,
delta_moving: [-0.1173993  -0.00045877 -0.0007931 ]
"""

So this decoupling of pos-orientation will lead to coupling of pos itself?

roberto-martinmartin, 2021-05-17:
Yes, this is right. It is better to not use the uncoupling. In this way, the full decoupling between position and orientation is used

closed 2021-06-21 by yukezhu