ARISE-Initiative/robosuite #227: How to connect with the algorithm in stable-baselines

caishanglei, 2021-05-09:
For example, action_space is not equivalent to gym.env. Is there an example of connecting with stable-baselines? Use the environment in robosuite

yukezhu, 2021-05-10:
This is beyond our support to the robosuite library. If you are looking for gym-like APIs, I suggest you look at the [GymWrapper](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/wrappers/gym_wrapper.py). Other than this, looking at the [robosuite-benchmark](https://github.com/ARISE-Initiative/robosuite-benchmark) repo would give you some idea on how to integrate robosuite with external RL libraries.

closed 2021-05-10 by yukezhu

hermanjakobsen, 2021-05-12:
Hi, @caishanglei! 

I have made a training script that uses PPO from [stable-baselines3](https://github.com/DLR-RM/stable-baselines3) to train a robosuite environment. The script is available [here](https://github.com/hermanjakobsen/robotic-ultrasound-imaging/blob/master/src/rl.py)

Hope it helps! :)