ARISE-Initiative/robosuite #237: How to disable collision checks?

SuryaPratapSingh37, 2021-06-28:
I am trying to attach a custom made gripper(similar to that shown in the figure) to the UR5e arm in robosuite but I feel there are collisions between the gripper and the arm making the arm make weird movements. The same weird behavior is not seen if there is an offset(greater than 3-4 cm) between the arm and gripper. So is there any way by which I can permanently disable collision checks between the arm and the gripper? Similarly, can i disable collision checks between the gripper and the conveyor belt since the gripper tries to push itself away from the conveyor belt when it goes pretty close to the conveyor belt which I guess is again because of collisions?
[image]

cremebrule, 2021-06-29:
Hi @SuryaPratapSingh37 ,

You can filter specific collisions by setting elements' `condim` and `conaffinity`, see HERE.

I would recommend against this, since you would have to manually hardcode values in the XMLs that are loaded (or create a new robot arm class with an edited XML file to specifically use with your custom gripper).

Instead, it sounds like your custom gripper is offset incorrectly with respect to the arm. You can toggle viewing the collision geoms by pressing `0` while viewing the `MjViewer` rendered screen at runtime, and toggle the visual geoms by pressing `1`. If it turns out your custom gripper has collision geoms overlapping with the arm, you can set the offset so that they're no longer in collision, e.g. [HERE](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/grippers/wiping_gripper.xml#L3) (this is for the Wiping Gripper, used as an example).

SuryaPratapSingh37, 2021-06-30:
Hi @cremebrule ,                                                                                                                                                                                         

Thank you for your reply, your suggestion about viewing the collision geoms was helpful since my custom gripper collision geom is indeed overlapping with the arm. 

Secondly, out of curiosity can you please also send some example for suppressing the pair of collisions using 'condim' and 'conaffinity' in the arm and gripper XMLs.

closed 2021-09-27 by yukezhu