ARISE-Initiative/robosuite #238: [Solve] ]Running into error when excute test_all_controllers.py

jdj2261, 2021-06-29:
I am currently trying to excute test_all_controllers.py in robosuite/tests/test_controllers directory.
however, I am running to the following error when doing so:

cannot find 'sawyer_description/meshes/base.DAE' in any directory in urdf path
cannot find 'sawyer_description/meshes/l5.STL' in any directory in urdf path

I modified sawyer_arm.urdf after moving to this path (robosuite/models/assets/bullet_data/sawyer_description/urdf)
After changing DAE and STL to lowercase, it was executed without error.

cremebrule, 2021-06-29:
Hi @jdj2261 ,

What version of robosuite are you on? This used to be an issue, but was fixed in one of our recent releases.

If you're using the current master, there shouldn't be a problem; the meshes already have [lowercased file extensions](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/models/assets/bullet_data/sawyer_description/meshes).

Closing this issue for now, feel free to reopen if you continue to have issues!

closed 2021-06-29 by cremebrule

jdj2261, 2021-06-30:
Hi @cremebrule,

I am currently using the master version.

Yes, [lowercased file extensions](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/models/assets/bullet_data/sawyer_description/meshes) doesn't matter.
It means [this file](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/bullet_data/sawyer_description/urdf/sawyer_arm.urdf) lines.

If you look at this lines(32, 52, 80, 107, 134, 161, 189, 216), you can see that it is capitalized.

Thanks for your reply.