ARISE-Initiative/robosuite #241: Calculation of torques in the osc.py control

danielstank, 2021-07-04:
In the osc.py the torques to be applied to the robot are given by the following expression
`self.torques = np.dot(self.J_full.T, decoupled_wrench) + self.torque_compensation`
Where `decoupled_wrench` is calculated as the output of PD controller.

I went to the Mujoco computation site (http://www.mujoco.org/book/computation.html) and I found the following: 
[image]

* the c: is equivalent to `self.torque_compensation`
* J^T*f: is  `np.dot(self.J_full.T, decoupled_wrench)`

BUT:
1) Is the multiplication of inertia matrix by the joint acceleration ignored because the desired acceleration should be zero? 
2) Why in the osc.py there is an addition between those terms unlike in the Mujoco documentation where there is substruction?

Thank you for your help,

roberto-martinmartin, 2021-07-11:
1. That equation is the simulated dynamics, not the control law. self.torques is our computation of "what should the torques be to achieve what we want". That is different to "what are the torques given all constraints" (what the simulator solves). At the end, in your controller with feedforward terms it all boils down to subtracting the effects that you can model in the best way. In this case, the inertia is usually very noisy (derivative of velocities in discrete timesteps) and it is supposed to have a negligible effect unless you are interested in very dynamic tasks like playing table tennis, or tossing objects. Also, the feedback loop would compensate for the unmodelled small disturbance.
2. It is the difference between the "plant model" and the "controller law": with the feedforward terms you are removing the effects you want to compensate for
3. For control? I tried using them for control but they are too noisy. As a generalization, simulators are accurate at producing realistic motion, even if the forces they generate are sometimes less realistic for that (spikes and transients...). If you just want to "read forces", there should be examples in the table wiping task or [here](https://github.com/ARISE-Initiative/robosuite/blob/18d9851b1d6500db654b0f941afa46e439a6e004/robosuite/robots/single_arm.py#L369)
Thanks

kaixindelele, 2021-07-12:
Hi, I am building a table tennis system and I want to control the robot arm to reach the specified position and attitude at the specified end speed and at the specified time based on "OSC_POSE" control. My current configuration is as follows.
```
{
    "type": "OSC_POSE",
    "input_max": 1,
    "input_min": -1,
    "output_max": [0.05, 0.05, 0.05, 0.5, 0.5, 0.5, 0.5],
    "output_min": [-0.05, -0.05, -0.05, -0.5, -0.5, -0.5],
    "kp": 150,
    "damping_ratio": 1,
    "impedance_mode": "fixed",
    "kp_limits": [0, 300],
    "damping_ratio_limits": [0, 10],
    "position_limits": null,
    "orientation_limits": null,
    "uncouple_pos_ori": true,
    "control_delta": true,
    "interpolation": null,
    "ramp_ratio": 0.2
}
```

When I have action=1, what is the end velocity unit of the robot arm and can I get its trajectory equation? Which parameters are related to it?

Your controller is too complex, looking forward to giving some directional guidance, thank you very much~

danielstank, 2021-07-12:
@roberto-martinmartin Thank you for your help

roberto-martinmartin, 2021-07-12:
what I do not recommend is direct force control: closing the loop with force/torque sensor readings. the impedance controller we provide does not do that, it only predicts the force applied with the end-effector
depends on what you want to do, but I think it would be hard to create a realistic control law with direct force measurements in simulation, or, at least, I would start with some in-depth work characterizing the FT measurements compared to the real-world ones and maybe implementing some filtering/noise/adaptation to bring them closer

closed 2021-07-17 by danielstank