ARISE-Initiative/robosuite #247: ft sensor rotatation and location relative to the gripper site

shirkozlovsky, 2021-08-11:
I noticed that there is a rotation between the ft sensor frame (that is located on the gripper link) to the gripper site (the site that is used for the control.
there is a reason for this rotation? 
also, I saw that the grip_site (the site that is used for the controller) is located below to sensor (0.145 in the z-direction ). it's not a problem for the sensor and the controller that not located in the same spot?

if I move the grip_site to the same rotation and location as the gripper link - it will work fine? or in could lead to some mistakes in the controller?
I'm working on UR5e and gripper 85

thank you

roberto-martinmartin, 2021-10-04:
Hi,

in what controller exactly? AFAIK, we do not provide any active force controller, we do not use the FT signals to compute controller response. If that is true, the FT sensor can be placed anywhere.
@cremebrule, is the grip_site the one used for control? I was expecting something like ee_site

@shirkozlovsky, I'm not sure where is the gripper link (or the reference frame of that link). What should happen is that the control is relative to that frame, e.g., if you query a pure rotation, it will be around the axes at the new frame location. If that is the behavior you prefer, that would work fine

cremebrule, 2022-02-23:
Hi @shirkozlovsky ,

Apologies for the (long) delay! To add to what roberto mentioned above:

- The FT sensor is attached to the "realistic" point on the robot -- i.e.: where the final robot arm link is attached to the gripper body.
- The grip site is the "actual" location of where grasping will / should occur. This results in more intuitive EEF control, where applying rotations occur about that grasping point (and therefore result in zero translation at that point). If we instead applied EEF control using the EEF frame as the target frame, this would result in both translation and rotation at the grasping point wrt to the world frame, which may be undesirable (and potentially harder to learn for an agent).

Closing this issue for now, feel free to reopen or start a new issue if you have additional questions / issues!

closed 2022-02-23 by cremebrule