ARISE-Initiative/robosuite #252: Wrong kinematical and dynamical properties for UR5e!!

EylonCohen, 2021-08-30:
I am building a simulation with the UR5e robot and I noticed that there is a difference between the data on the company website and the robosuite XML file. I think that in the XML file the data belongs to the UR5 robot and not to the UR5e.

I would appreciate any response to this issue!

website link: 
https://www.universal-robots.com/articles/ur/application-installation/dh-parameters-for-calculations-of-kinematics-and-dynamics/

roberto-martinmartin, 2021-10-03:
Hi,
I haven't personally used Universal Robots but the information I found about it is that there are no changes in sizes/kinematics between both, just sensing and some new programming interfaces:
https://www.robots.com/robots/universal-robots-ur5e
"With the same specs as the UR5, this robot has been improved by adding a Force Torque Sensor in Axis 6 of the robot. It also has a HMI with Capacitive touchscreen and higher resolution. The other benefit of this newer UR5e robot is programming and navigating improvement."

Would you mind sharing the XML files you think are the correct ones for UR5 and UR5e? We would be happy to correct any mistakes to facilitate sim2real.

Thanks!

EylonCohen, 2021-10-08:
Hi, thanks for replying.

Although the link you add to your comment says that the two robot versions are the same, you can see in the specific dynamical and kinematical spec I add that they are not.

Unfortunately, I don't have any updated XML for the ur5e. 

[image]
[image]

closed 2023-03-19 by yukezhu