ARISE-Initiative/robosuite #256: Speeding up off-screen simulation

garlicbutter, 2021-09-25:
Hi,
We are currently running 1000+ episodes of simulation with different parameters.
All of the simulations are run off-screen to speed up the process.
However, one episode still takes about 20 seconds to finish.
We were wondering if there's a way to speed up the off-screen simulation.

Our environment arguments are as follows:
        use_camera_obs=False,
        use_object_obs=True,
        has_renderer=False,
        has_offscreen_renderer=False,
        control_freq=20,
        ignore_done=True,
        hard_reset=True

yukezhu, 2021-09-27:
Hi @garlicbutter how many action steps are there in each of the episodes? Our simulation runs in the range of 20-100 steps per second. When you said it takes 20 seconds to finish, it really depends on how long each episode is.

garlicbutter, 2021-09-28:
It takes about 400 steps. So I suppose that's a reasonable run time.  So does the controller frequency determine how many actions are taken per second in real time? The time between running with screen renderer and running with off-screen has no major difference. I was wondering if ROBOSUITE takes real time into account.

yukezhu, 2021-10-04:
> So does the controller frequency determine how many actions are taken per second in real time?

Yes, that's correct.

closed 2021-10-04 by yukezhu