ARISE-Initiative/robosuite #257: glfw window viewer application not responding on macos

shivanimall, 2021-10-04:
Hello, 

I am on macos, I can successfully run with robosuite, but am unable to see the viewer. 

referred to this to load on-screen: https://robosuite.ai/docs/modules/environments.html:
```
# create an environment to visualize on-screen
env = robosuite.make(
    "TwoArmLift",
    robots=["Sawyer", "Panda"],             # load a Sawyer robot and a Panda robot
    gripper_types="default",                # use default grippers per robot arm
    controller_configs=controller_config,   # each arm is controlled using OSC
    env_configuration="single-arm-opposed", # (two-arm envs only) arms face each other
    has_renderer=True,                      # on-screen rendering
    render_camera="frontview",              # visualize the "frontview" camera
    has_offscreen_renderer=False,           # no off-screen rendering
    control_freq=20,                        # 20 hz control for applied actions
    horizon=200,                            # each episode terminates after 200 steps
    use_object_obs=False,                   # no observations needed
    use_camera_obs=False,                   # no observations needed
)

```
To note, I have successfully opened igibson, mujoco and pybullet viewers.

Following is the output:
`/Users/shivanimall/robosuite/robosuite
<module 'robosuite' from '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/robosuite-1.2.2-py3.9.egg/robosuite/__init__.py'>
Creating window glfw
2021-10-04 13:11:35.262 Python[97887:52053130] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/fd/5jk_ll1j7gzbjtr57ltsdtww0000gn/T/org.python.python.savedState
Creating window glfw
`
but the window instance says application not responding.

yukezhu, 2021-10-31:
Dear @shivanimall, it appears to me that this issue was not caused by robosuite, but some configurations of your system runtime. I can confirm that the mjc_viewer works fine on my macOS (I have Big Sur). I would suggest you search with the error message "ApplePersistenceIgnoreState: Existing state will not be touched." and see if you can find any online solutions that resolve your problem.

closed 2021-10-31 by yukezhu

shivanimall, 2021-11-05:
thanks @yukezhu! want to confirm that mujoco viewer is now working.