ARISE-Initiative/robosuite #267: Gym Wrapper and Rewards

sohams-MASS, 2021-11-11:
When implementing the GymWrapper on an Robosuite Environment , the rewards obtained from env.step(action) is always 0 - just wondering if I need to redefine the rewards function in the GymWrapper? 

or should the rewards obtained from the GymWrapper be the same as the rewards obtained from RoboSuite Environment?

cremebrule, 2022-01-08:
Hi @sohams-MASS ,

Apologies for the delay in response. The rewards should be identical to the underlying environment being wrapped. Which environment are you using?

cremebrule, 2022-02-23:
Hi @sohams-MASS ,

Since there hasn't been a response for awhile, I'm going to close this issue for now. Feel free to reopen if you have additional questions / problems!

closed 2022-02-23 by cremebrule