ARISE-Initiative/robosuite #269: TypeError with xml.etree.ElementTree import

Gongkaka, 2021-12-03:
I followed the installation instructions and try the demo_random_action.py, then got this error:

```
File "/home/gxxxa/packages/robosuite-master/robosuite/models/base.py", line 185, in merge_assets
    self.asset.append(asset)
TypeError: append() argument must be xml.etree.ElementTree.Element, not Element
```

yukezhu, 2021-12-04:
Hi @Gongkaka did you install robosuite through pip or using the git clone? I tried a clean install and wasn't able to reproduce your issue. More information would be necessary for us to investigate this further.

Gongkaka, 2021-12-06:
> Hi @Gongkaka did you install robosuite through pip or using the git clone? I tried a clean install and wasn't able to reproduce your issue. More information would be necessary for us to investigate this further.

Hi Prof. Zhu. I just realized that I need to install _numpy_ in my virtual environment as well..Sorry for wasting two minutes of your life lol.

closed 2021-12-06 by Gongkaka

lorepieri8, 2022-02-17:
I got the same issue, and fixed it by installing the requirements in a fresh venv environment.