ARISE-Initiative/robosuite #270: Question on OSC vs IK controllers

hueds, 2021-12-05:
I am reading the module documentation on [controllers](https://robosuite.ai/docs/modules/controllers.html), and it seems to have some stale or conflicting information on the choice of controllers.

The controller documentation mentions operational-space control, but not IK in the first paragraph. 
> Our controller options include OSC_POSE, OSC_POSITION, JOINT_POSITION, JOINT_VELOCITY, and JOINT_TORQUE.

It then briefly mentions IK in the third paragraph. 
> When using any position-based control (OSC, IK, or JOINT_POSITION controllers), input actions are, by default, interpreted as delta values from the current state.

The following paragraphs that detail all the controllers do not mention IK at all. 

When I read the Simulation API's documentation on the [controller](https://robosuite.ai/docs/simulation/controller.html#inverse-kinematics-controller), there is more information on the IK. 

Can the authors compare and contrast the two controllers? It seems like both enable positional and orientational control of the gripper. It would also be good if an additional paragraph about IK could be added to the module documentation of controllers as well.

hueds, 2021-12-07:
I was able to find some more information in the [docs](https://robosuite.ai/docs/demos.html) about the differences.
> OSC_POSE: Gripper moves sequentially and linearly in x, y, z direction, then sequentially rotates in x-axis, y-axis, z-axis, relative to the global coordinate frame
>
> OSC_POSITION: Gripper moves sequentially and linearly in x, y, z direction, relative to the global coordinate frame

cremebrule, 2022-01-08:
Hi @hueds ,

Apologies for the delay in response. You're correct in that both OSC and IK achieve the same end goal: desired end-effector cartesian movement, but the method of achieving this motion is quite different.

The key difference is that IK is a purely kinematic controller -- that is, it only considers the _geometric_ constraints imposed by the robot (i.e.: how long each robot link is, what the joints are, etc.). Because of this, it cannot take into account any dynamics, such as impedances (e.g.: someone pushing against the arm during movement).

In contrast, OSC is a _dynamics_ based controller -- it takes into account impedance "gains" in each of the 6 cartesian degrees of freedom (x,y,z translation, and rx,ry,rz rotation), and produces motion corresponding to these gains. Thus, OSC can be useful for applications such as contact-rich motion, which requires motion amidst external forces.

In general, OSC is a much more expressive and powerful controller, but in practice can be much more difficult to deploy due to its sensitivity on modeling parameters (i.e.: the time-varying mass matrix).

@roberto-martinmartin can chime in with further info.

Thanks for bringing up the lack of IK documentation in our docs! We will address this in a future release.

hueds, 2022-01-09:
Thank you, this was very helpful.

closed 2022-01-09 by hueds