ARISE-Initiative/robosuite #272: lowercasing of DAE and STL file extension in robot URDF necessary for IK control

hueds, 2021-12-07:
The assets have lowercased `.DAE`, `.STL` files extensions, but the URDF has capitalized `.DAE` and `.STL` file extensions. This causes a pybullet error on loading the URDF for the IK_POSE controller. To fix, the URDF needs to be updated.   

This was referenced in a previous issue:

It means [this file](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/bullet_data/sawyer_description/urdf/sawyer_arm.urdf) lines.

If you look at this lines(32, 52, 80, 107, 134, 161, 189, 216), you can see that it is capitalized.

Thanks for your reply.

_Originally posted by @jdj2261 in https://github.com/ARISE-Initiative/robosuite/issues/238#issuecomment-871080113_

yukezhu, 2021-12-28:
@awesome-aj0123 can you make a PR to address this?

yukezhu, 2022-01-10:
Thank you for reporting @hueds ! This should be fixed now.

closed 2022-01-10 by yukezhu