ARISE-Initiative/robosuite #275: expected FPS

yiwc, 2021-12-22:
Dear Robosuite team,

Thanks for your great work in robosuite! May I know what's the expected env iteration fps? with image render? I have tried a demo, and found the fps is about 8, which is quite low that I think. Is it fps=8 a normal case? or I wrongly setting sth?

Much THanks

yiwc, 2021-12-22:
`import robosuite as suite

render_on_screen=False
assert render_on_screen==False

import rl_helper
from FIRL.play.play_ray.wrappers import RobosuiteRGBImgPartialObsWrapper,RobosuiteImgObsFloatNormalizeWrapper
from gym import spaces
if __name__=="__main__":=
    env_name="PickPlaceCereal"
    robot_name="Panda"
    env = suite.make(
        env_name=env_name, # try with other tasks like "Stack" and "Door"
        robots=robot_name,  # try with other robots like "Sawyer" and "Jaco"
        has_renderer=render_on_screen,
        has_offscreen_renderer=~render_on_screen,
        render_gpu_device_id=3,
        use_camera_obs=True,
        render_camera="agentview",
        renderer="mujoco"
    )

    env = RobosuiteRGBImgPartialObsWrapper(env)
    env = RobosuiteImgObsFloatNormalizeWrapper(env)
    rl_helper.fps(env)`

yukezhu, 2021-12-28:
fps=8 sounds quite low. The exact fps depends on many factors, such as scene complexity, renderers, hardware, etc. From our [Renderers](https://robosuite.ai/docs/modules/renderers.html) page, you can see some profiling results that we obtained from running robosuite on our hardware with the renderers.

closed 2022-02-23 by yukezhu