ARISE-Initiative/robosuite #276: Add rope object

mgualti, 2021-12-29:
Is it possible to add a rope-like object to the environment? Combining the example XML for a rope on this page (https://mujoco.readthedocs.io/en/latest/modeling.html#loading-models) with the example XML on this page (https://robosuite.ai/docs/modules/objects.html), I get the following:

<mujoco>
    <worldbody>

        <body>
            <body name="object" pos="0 0 1">
                <freejoint />
                <composite type="rope" count="21 1 1" spacing="0.04" offset="0 0 2">
                    <joint kind="main" damping="0.005" />
                    <geom type="capsule" size=".01 .015" rgba=".8 .2 .1 1" group="10" />
                </composite>
            </body>

            <site rgba="0 0 0 0" size="0.005" pos="0 0 -0.045" name="bottom_site" />
            <site rgba="0 0 0 0" size="0.005" pos="0 0 0.03" name="top_site" />
            <site rgba="0 0 0 0" size="0.005" pos="0.03 0.03 0" name="horizontal_radius_site" />
        </body>

    </worldbody>
</mujoco>

But it crashes with a message complaining about the object name not starting with a "B". Is there a better way to add a rope?

cremebrule, 2022-01-08:
Hi @mgualti ,

We unfortunately do not officially support composite mujoco objects currently. The main issue is that these composite objects are not "fully" specified in the XML (e.g.: the one you posted above), and are, at runtime, interpreted and expanded into the full set of bodies / geoms. i.e.: that `<composite>` tag ends up being 21 chained geoms. Our current infra isn't built to support this feature yet.

However, we do have internal efforts exploring these types of composite objects. @yukezhu , can we perhaps link your student that was working with rope in robosuite with @mgualti to provide further insight?

yukezhu, 2022-01-10:
Hi @mgualti feel free to reach out to [@JieFeng-c](https://github.com/JieFeng-cse) He has worked on modeling deformable objects in the past. He might be able to give you some further pointers.

closed 2022-01-10 by yukezhu