ARISE-Initiative/robosuite #282: robotiq 85 xml

guodashun, 2022-01-13:
Dear Robosuite team,
Thanks for your great work in robosuite! I'm trying to add robotiq 2f-85 model to my env and I've found there is a [perpared xml](https://github.com/ARISE-Initiative/robosuite/blob/56c20db6231e03ec5910869f7227819943fc24b2/robosuite/models/assets/grippers/robotiq_gripper_85.xml) model of this gripper. But when I load this xml I meet some problems.

```
<mesh name="robotiq_arg2f_85_outer_knuckle" file="meshes/robotiq_85_gripper/robotiq_arg2f_85_outer_knuckle.stl" scale="0.001 0.001 0.001" />
<mesh name="robotiq_arg2f_85_outer_finger" file="meshes/robotiq_85_gripper/robotiq_arg2f_85_outer_finger.stl" scale="0.001 0.001 0.001" />
<mesh name="robotiq_arg2f_85_inner_finger" file="meshes/robotiq_85_gripper/robotiq_arg2f_85_inner_finger.stl" scale="0.001 0.001 0.001" />
<mesh name="robotiq_arg2f_85_inner_knuckle" file="meshes/robotiq_85_gripper/robotiq_arg2f_85_inner_knuckle.stl" scale="0.001 0.001 0.001" />
<mesh name="robotiq_arg2f_base_link" file="meshes/robotiq_85_gripper/robotiq_arg2f_base_link_vis.stl" />
```

In the line 4 and 8, there is an same mesh name ***<robotiq_arg2f_base_link>***.
And in the ***mesh*** folder, I could not find the file **robotiq_arg2f_base_link_vis.stl**.
Then I run the demo [test_gripper](https://github.com/ARISE-Initiative/robosuite/blob/56c20db6231e03ec5910869f7227819943fc24b2/robosuite/demos/demo_gripper_interaction.py), I've found the robotiq85 can actually work.
It's confusing to me, maybe there are some settings missing for me?
Thank you very much.

cremebrule, 2022-02-23:
Hi @guodashun ,

How are you trying to load the XML? There could be a potential issue due to the fact that the meshes are specified as relative paths, rather than absolute paths.

For reference, all the robotiq 85 mesh files are found [HERE](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/models/assets/grippers/meshes/robotiq_85_gripper). Can you try specifying absolute paths instead and seeing if that fixes your problem?

guodashun, 2022-03-02:
Thanks for your reply. The problem is that I could not find the **_robotiq_arg2f_base_link_vis.stl_** in the given link.

cambel, 2023-11-17:
@guodashun, you are right, that mesh is missing from the repository.