ARISE-Initiative/robosuite #290: Issue with loading XML objects

michaal94, 2022-02-02:
I am trying to build my environment, however, I cannot seem to load any XML based object. The following minimal example:
```
from robosuite.models import MujocoWorldBase
from robosuite.models.objects import MilkObject

world = MujocoWorldBase()
milk = MilkObject('milk').get_obj()
world.worldbody.append(milk)
model = world.get_model(mode="mujoco_py")
```
results in:
```
Traceback (most recent call last):
  File "test2.py", line 7, in <module>
    model = world.get_model(mode="mujoco_py")
  File "/home/michas/anaconda3/envs/pytorch/lib/python3.7/site-packages/robosuite/models/base.py", line 133, in get_model
    model = load_model_from_xml(string.getvalue())
  File "anaconda3/envs/pytorch/lib/python3.7/site-packages/mujoco_py/cymj.pyx", line 191, in mujoco_py.cymj.load_model_from_xml
Exception: <mujoco model="base">
  <compiler angle="radian" inertiagrouprange="0 0" meshdir="meshes/" />
  <option cone="elliptic" density="1.2" impratio="20" timestep="0.002" viscosity="0.00002" />
  <size nconmax="5000" njmax="5000" />
  <asset>
  </asset>
  <visual>
    <map znear="0.001" />
  </visual>
  <actuator>
  </actuator>
  <worldbody>
  <body name="milk_main">
        <geom condim="4" density="100" friction="0.95 0.3 0.1" group="0" mesh="milk_mesh" name="milk_g0" pos="0 0 0" rgba="0.5 0 0 1" solimp="0.998 0.998 0.001" solref="0.001 1" type="mesh" />
      <geom conaffinity="0" condim="4" contype="0" density="100" friction="0.95 0.3 0.1" group="1" mass="1e-8" material="milk_ceramic" mesh="milk_mesh" name="milk_g0_visual" pos="0 0 0" solimp="0.998 0.998 0.001" solref="0.001 1" type="mesh" />
      <joint damping="0.0005" name="milk_joint0" type="free" /><site group="0" name="milk_default_site" pos="0 0 0" rgba="1 0 0 0" size="0.002 0.002 0.002" type="sphere" /></body>
      </worldbody>
  <equality>
  </equality>
<sensor /><tendon /><contact /></mujoco>
Failed to load XML from string. mj_loadXML error: b"Error: mesh 'milk_mesh' not found in geom 0\nObject name = milk_g0, id = 0, line = 18, column = 9"
```

cremebrule, 2022-02-23:
Hi @michaal94,

The issue is that the when you added the milk object manually via `.append(...)`, it doesn't automatically port over the assets and other dependencies from the original XMLObject. The intended API you want to use is `merge`:

```python
world = MujocoWorldBase()
milk = MilkObject('milk')           # Note how we don't call .get_obj()!
world.merge(milk)
```

Closing this issue for now, feel free to re-open if you have additional issues!

closed 2022-02-23 by cremebrule

kevin-thankyou-lin, 2022-05-01:
Following on this thread, how do you set the positions of these XML objects (I know there're the `Sampler` classes, but how would I manually set positions)?

Also, I can't seem to see the milk (or any other XML object) after using the `world.merge(...)` command.

Script to reproduce:
```
from robosuite.models.objects import BallObject
from robosuite.models.objects import CylinderObject, CerealObject, BreadObject, MilkObject
from robosuite.utils.mjcf_utils import new_joint

from robosuite.models.arenas import TableArena

from robosuite.models import MujocoWorldBase

world = MujocoWorldBase()

mujoco_arena = TableArena()
mujoco_arena.set_origin([0.8, 0, 0])
world.merge(mujoco_arena)
milk = MilkObject('milk')           # Note how we don't call .get_obj()!
world.merge(milk)

world.merge(cereal)

model = world.get_model(mode="mujoco_py")

from mujoco_py import MjSim, MjViewer

sim = MjSim(model)
viewer = MjViewer(sim)  
viewer.vopt.geomgroup[0] = 0 # disable visualization of collision mesh

for i in range(10000):
    sim.step()
    viewer.render()
```

kevin-thankyou-lin, 2022-05-01:
Ah, resolve the issue by removing `viewer.vopt.geomgroup[0] = 0 # disable visualization of collision mesh`