ARISE-Initiative/robosuite #292: Reset robot to desired end-effector positions

xuanhien070594, 2022-02-08:
Hi,

I'm trying to figure out how to set the robot to desired end-effector positions during initialization of an environment, but I couldn't. May I know if that is possible with robosuite right now? Thanks!

cremebrule, 2022-02-23:
Hi @xuanhien070594 ,

Does the arm configuration matter or do you only care about the EEF position? If the latter, then you could automatically find a valid arm configuration that reaches the desired position by controlling the robot EEF to the desired position (e.g.: using OSC POSE / POSITION), and then grabbing the robot's joints at that configuration, and then hardcoding it in your robot via, e.g.: `env.robots[0].init_qpos = <DESIRED_JOINT_CONFIG>`

xuanhien070594, 2022-03-01:
Many thanks for your reply!

closed 2022-03-01 by xuanhien070594