ARISE-Initiative/robosuite #293: robosuite v1.1 ValueError: numpy.ndarray size changed, may indicate binary incompatibility.

1172534699, 2022-02-14:
Dear Robosuite team,

Thanks for your great work in robosuite! i am a student. I want to run the MAPLE ,but when i installed my envirenment MAPLE,i found that my numpy version is incompatibility. But I install the env  MAPLE followed requirements.txt(exactly maple.yml),so what is wrong?

Xia Xiaowu

1172534699, 2022-02-14:
(maple) summer@xxw-OMEN:~/robosuite_maple$ python robosuite/demos/demo_random_action.py 
Traceback (most recent call last):
  File "robosuite/demos/demo_random_action.py", line 1, in <module>
    from robosuite.controllers import load_controller_config
  File "/home/summer/robosuite_maple/robosuite/__init__.py", line 1, in <module>
    from robosuite.environments.base import make
  File "/home/summer/robosuite_maple/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/summer/robosuite_maple/robosuite/environments/base.py", line 2, in <module>
    from mujoco_py import MjSim, MjRenderContextOffscreen
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/mujoco_py/builder.py", line 510, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/mujoco_py/builder.py", line 101, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/mujoco_py/builder.py", line 125, in load_dynamic_ext
    return loader.load_module()
  File "mujoco_py/cymj.pyx", line 1, in init mujoco_py.cymj
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 80 from PyObject

1172534699, 2022-02-14:
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
absl-py                   0.12.0                    <pip>
addict                    2.4.0                     <pip>
anyio                     3.3.1                     <pip>
argon2-cffi               20.1.0           py38h27cfd23_1  
astunparse                1.6.3                     <pip>
async_generator           1.10               pyhd3eb1b0_0  
attrs                     21.4.0             pyhd3eb1b0_0  
attrs                     20.2.0                    <pip>
Babel                     2.9.1                     <pip>
backcall                  0.2.0              pyhd3eb1b0_0  
bleach                    3.3.0              pyhd3eb1b0_0  
ca-certificates           2021.1.19            h06a4308_1  
cachetools                4.2.1                     <pip>
certifi                   2020.12.5        py38h06a4308_0  
cffi                      1.14.2                    <pip>
cffi                      1.15.0           py38hd667e15_1  
chardet                   3.0.4                     <pip>
clang                     5.0                       <pip>
click                     7.1.2                     <pip>
cloudpickle               1.3.0                     <pip>
cycler                    0.10.0                    <pip>
Cython                    0.29.21                   <pip>
dbus                      1.13.18              hb2f20db_0  
decorator                 4.4.2              pyhd3eb1b0_0  
defusedxml                0.7.1              pyhd3eb1b0_0  
deprecation               2.1.0                     <pip>
dill                      0.3.4                     <pip>
entrypoints               0.3                      py38_0  
expat                     2.3.0                h2531618_2  
fasteners                 0.15                      <pip>
Flask                     1.0.2                     <pip>
flatbuffers               1.12                      <pip>
fontconfig                2.13.1               h6c09931_0  
freetype                  2.10.4               h5ab3b9f_0  
future                    0.18.2                    <pip>
gast                      0.4.0                     <pip>
gitdb                     4.0.2                     <pip>
gitdb2                    2.0.6                     <pip>
GitPython                 2.1.7                     <pip>
glfw                      1.12.0                    <pip>
glib                      2.68.0               h36276a3_0  
google-api-core           1.26.0                    <pip>
google-api-python-client  1.12.8                    <pip>
google-auth               1.26.1                    <pip>
google-auth-httplib2      0.0.4                     <pip>
google-auth-oauthlib      0.4.4                     <pip>
google-cloud-core         1.6.0                     <pip>
google-cloud-storage      1.36.0                    <pip>
google-crc32c             1.1.2                     <pip>
google-pasta              0.2.0                     <pip>
google-resumable-media    1.2.0                     <pip>
googleapis-common-protos  1.52.0                    <pip>
grpcio                    1.39.0                    <pip>
gst-plugins-base          1.14.0               h8213a91_2  
gstreamer                 1.14.0               h28cd5cc_2  
gtimer                    1.0.0b5                   <pip>
gym                       0.17.2                    <pip>
h5py                      3.1.0                     <pip>
httplib2                  0.19.0                    <pip>
icu                       58.2                 he6710b0_3  
idna                      2.10                      <pip>
imageio                   2.9.0                     <pip>
imageio-ffmpeg            0.4.4                     <pip>
importlib-metadata        3.7.3            py38h06a4308_1  
importlib_metadata        3.7.3                hd3eb1b0_1  
ipykernel                 5.3.4            py38h5ca1d4c_0  
ipython                   7.22.0           py38hb070fc8_0  
ipython_genutils          0.2.0              pyhd3eb1b0_1  
[... 125 lines cut ...]
testpath                  0.4.4              pyhd3eb1b0_0  
threadpoolctl             2.2.0                     <pip>
tifffile                  2021.8.8                  <pip>
tk                        8.6.10               hbc83047_0  
tomlkit                   0.7.2                     <pip>
torch                     1.6.0                     <pip>
torchvision               0.7.0                     <pip>
tornado                   6.1              py38h27cfd23_0  
tqdm                      4.61.1                    <pip>
traitlets                 5.0.5              pyhd3eb1b0_0  
typing-extensions         3.7.4.3                   <pip>
uritemplate               3.0.1                     <pip>
urllib3                   1.25.10                   <pip>
wcwidth                   0.2.5                      py_0  
webencodings              0.5.1                    py38_1  
websocket-client          1.2.1                     <pip>
Werkzeug                  1.0.1                     <pip>
wheel                     0.37.1             pyhd3eb1b0_0  
wheel                     0.37.0                    <pip>
widgetsnbextension        3.5.1                    py38_0  
wrapt                     1.12.1                    <pip>
xz                        5.2.5                h7b6447c_0  
zeromq                    4.3.4                h2531618_0  
zipp                      3.4.1              pyhd3eb1b0_0  
zlib                      1.2.11               h7b6447c_3

snasiriany, 2022-02-14:
Hi Xia,

Thanks for bringing this up to our attention. Seems like other `mujoco_py` users have had a similar issue: https://github.com/openai/mujoco-py/issues/607

That thread suggests three possible solutions:
1. Change numpy to version 1.20.0: `pip install numpy==1.20.0`
2. Change mujoco_py to version 2.0.2.5: `pip install mujoco_py==2.0.2.5`
3. Another user also suggested: `pip install mujoco-py==2.0.2.9 --no-cache-dir --no-binary :all: --no-build-isolation`

Do any of these help solve the issue?

Soroush

1172534699, 2022-02-16:
Oh,yes! I solved it by your third suggest(pip install mujoco-py==2.0.2.9 --no-cache-dir --no-binary :all: --no-build-isolation),thank you very much! Thank you for answering me!You are a good man haha! But it comes a new bug that is TypeError: 'NoneType' object is not iterable,I think it is also caused by env or a small mistake.Did you encounter this problem? Hope for your answer !

(maple) summer@xxw-OMEN:~/robosuite_maple$ python robosuite/demos/demo_random_action.py
Welcome to robosuite v1.1.0!

      ;     /        ,--.
     ["]   ["]  ,<  |__**|
    /[_]\  [~]\/    |//  |
     ] [   OOO      /o|__|

Here is a list of environments in the suite:

[0] Cleanup
[1] Door
[2] Lift
[3] NutAssembly
[4] NutAssemblyRound
[5] NutAssemblySingle
[6] NutAssemblySquare
[7] PegInHole
[8] PickPlace
[9] PickPlaceBread
[10] PickPlaceCan
[11] PickPlaceCereal
[12] PickPlaceMilk
[13] PickPlaceSingle
[14] Stack
[15] Wipe

Choose an environment to run (enter a number from 0 to 15): 0
Here is a list of available robots:

[0] IIWA
[1] Jaco
[2] Kinova3
[3] Panda
[4] Sawyer
[5] UR5e

Choose a robot (enter a number from 0 to 5): 2
Here is a list of controllers in the suite:

[0] JOINT_VELOCITY - Joint Velocity
[1] JOINT_TORQUE - Joint Torque
[2] JOINT_POSITION - Joint Position
[3] OSC_POSITION - Operational Space Control (Position Only)
[4] OSC_POSE - Operational Space Control (Position + Orientation)
[5] OSC_POSITION_YAW - Operational Space Control (Position + Yaw)
[6] IK_POSE - Inverse Kinematics Control (Position + Orientation) (Note: must have PyBullet installed)

Choose a controller for the robot (enter a number from 0 to 6): 2

Press "H" to show the viewer control panel.
Traceback (most recent call last):
  File "robosuite/demos/demo_random_action.py", line 50, in <module>
    env = suite.make(
  File "/home/summer/robosuite_maple/robosuite/environments/base.py", line 41, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "/home/summer/robosuite_maple/robosuite/environments/manipulation/cleanup.py", line 188, in __init__
    super().__init__(
  File "/home/summer/robosuite_maple/robosuite/environments/manipulation/manipulation_env.py", line 148, in __init__
    super().__init__(
  File "/home/summer/robosuite_maple/robosuite/environments/robot_env.py", line 167, in __init__
    self.skill_controller = SkillController(self, skill_config)
  File "/home/summer/robosuite_maple/robosuite/controllers/skill_controller.py", line 29, in __init__
    self._setup_config(config)
  File "/home/summer/robosuite_maple/robosuite/controllers/skill_controller.py", line 46, in _setup_config
    self._config.update(config)
TypeError: 'NoneType' object is not iterable

1172534699, 2022-02-16:
And if I change my mujoco_py to version 2.0.2.5,it faces the same question.(I'm a novice)

snasiriany, 2022-02-17:
Hi Xia,

Great to hear that the mujoco_py issue was resolved. Regarding the new issue that you mentioned -- that script `demo_random_action.py` hasn't been updated to work with the MAPLE framework, so the error is expected. In fact many of the programs under `robosuite/demos` and `robosuite/scripts` are probably stale and are not compatible with MAPLE. In the future I will address these issues. For now, `robosuite/scripts/collect_human_demonstrations.py` does work, and it allows you to interactively play with the environment. Let me know if this specific script brings any issues for you.

Soroush

1172534699, 2022-02-18:
Dear Soroush,
I am very glad to hear from you.Thank you for your patience and thoughtful answer.I had known what you expressed.So I run the code robosuite/scripts/collect_human_demonstrations.py ,it does work,which verified that my _robosuite_ environment is OK.
To be honest,I'm a 1st year master student and interested in your paper _Augmenting Reinforcement Learning with Behavior Primitives for Diverse Manipulation Tasks_ . So I want to run it,But I still encounter a question.I had tried to solve it. I see _stochastic_ file in _maple/torch/networks_ but why in the _build/lib/maple/torch/networks_ I can't find _stochastic_, I followed your README.md to install _maple==0.2.1.dev0_ .Is there something wrong with me. Maybe I still need your help.
XIA xiaowu

(maple) summer@xxw-OMEN:~/maple$ python scripts/train.py --env stack
No personal conf_private.py found.
doodad not detected
Traceback (most recent call last):
  File "scripts/train.py", line 4, in <module>
    from maple.launchers.robosuite_launcher import experiment
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/maple-0.2.1.dev0-py3.8.egg/maple/launchers/robosuite_launcher.py", line 8, in <module>
    from maple.torch.sac.policies import (
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/maple-0.2.1.dev0-py3.8.egg/maple/torch/sac/policies/__init__.py", line 1, in <module>
    from maple.torch.sac.policies.base import (
  File "/home/summer/anaconda3/envs/maple/lib/python3.8/site-packages/maple-0.2.1.dev0-py3.8.egg/maple/torch/sac/policies/base.py", line 17, in <module>
    from maple.torch.networks.stochastic.distribution_generator import (
ModuleNotFoundError: No module named 'maple.torch.networks.stochastic'

snasiriany, 2022-02-18:
Did you try `pip install -e .` in the root directory of this repository? This is important. If you do this, I think everything will work. But if you run `pip maple==0.2.1.dev0` I don't think this will work. So I'd suggest uninstalling the maple repo, and reinstalling by running `pip install -e .` from the **root folder** of the maple repo.

closed 2022-04-03 by yukezhu

MrGZC, 2023-07-05:
> Dear Robosuite team,

Dear Robosuite team,

I have tried the three ways you have suggested. However, I need help solving the problem. The output when I perform the order "env = gym.make('Safexp-PointGoal1-v0')" is 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/gym/envs/registration.py", line 156, in make
    return registry.make(id, **kwargs)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/gym/envs/registration.py", line 101, in make
    env = spec.make(**kwargs)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/gym/envs/registration.py", line 72, in make
    cls = load(self.entry_point)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/gym/envs/registration.py", line 17, in load
    mod = importlib.import_module(mod_name)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/user/gzc3090/code/safety-gym/safety_gym/envs/mujoco.py", line 9, in <module>
    from safety_gym.envs.engine import *  # noqa
  File "/home/user/gzc3090/code/safety-gym/safety_gym/envs/engine.py", line 9, in <module>
    import mujoco_py
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/mujoco_py/__init__.py", line 3, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/mujoco_py/builder.py", line 506, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/mujoco_py/builder.py", line 101, in load_cython_ext
    mod = load_dynamic_ext('cymj', cext_so_path)
  File "/home/user/gzc3090/anaconda3/envs/safetygym/lib/python3.8/site-packages/mujoco_py/builder.py", line 125, in load_dynamic_ext
    return loader.load_module()
  File "mujoco_py/cymj.pyx", line 1, in init mujoco_py.cymj
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 96 from C header, got 80 from PyObject

==============================================================
When I tried " pip install numpy==1.20.0", the output is
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
safety-gym 0.0.0 requires numpy~=1.17.4, but you have numpy 1.20.0 which is incompatible.
==============================================================

The other two orders can't work well also. And the packages I installed  are:
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main
_openmp_mutex             5.1                       1_gnu
ca-certificates           2023.05.30           h06a4308_0
cffi                      1.15.1                   pypi_0    pypi
cloudpickle               1.2.2                    pypi_0    pypi
cython                    0.29.36                  pypi_0    pypi
fasteners                 0.18                     pypi_0    pypi
future                    0.18.3                   pypi_0    pypi
glfw                      2.6.2                    pypi_0    pypi
gym                       0.15.7                   pypi_0    pypi
imageio                   2.31.1                   pypi_0    pypi
joblib                    0.14.1                   pypi_0    pypi
ld_impl_linux-64          2.38                 h1181459_1
libffi                    3.3                  he6710b0_2
libgcc-ng                 11.2.0               h1234567_1
libgomp                   11.2.0               h1234567_1
libstdcxx-ng              11.2.0               h1234567_1
mujoco-py                 2.0.2.7                  pypi_0    pypi
ncurses                   6.4                  h6a678d5_0
numpy                     1.17.5                   pypi_0    pypi
openssl                   1.1.1u               h7f8727e_0
patchelf                  0.17.2.1                 pypi_0    pypi
pillow                    10.0.0                   pypi_0    pypi
pip                       23.1.2           py38h06a4308_0
pycparser                 2.21                     pypi_0    pypi
pyglet                    1.5.0                    pypi_0    pypi
python                    3.8.13               haa1d7c7_1
readline                  8.2                  h5eee18b_0
safety-gym                0.0.0                     dev_0    <develop>
scipy                     1.8.1                    pypi_0    pypi
setuptools                67.8.0           py38h06a4308_0
six                       1.16.0                   pypi_0    pypi
sqlite                    3.41.2               h5eee18b_0
tk                        8.6.12               h1ccaba5_0
wheel                     0.38.4           py38h06a4308_0
xmltodict                 0.12.0                   pypi_0    pypi
xz                        5.4.2                h5eee18b_0
zlib                      1.2.13               h5eee18b_0

I hope to receive your reply, best, I wish you the best of luck.