ARISE-Initiative/robosuite #294: Parallel learning using robosuite and stable-baselines3

danielstankw, 2022-02-15:
I have tried using the SubVecProcEnv for parallel learning using stable-baselines3 and PPO but I noticed not learning speed improvement. Did anyone succeed in speeding up the learning and could share how they did it?

cremebrule, 2022-02-23:
Hi @danielstankw ,

Thanks for bringing up this question. Can you share some more details about your experiment setup -- i.e.: what task you're running, horizon, training config, how long it took to train? IIRC when I was trying PPO it took only a few hours to train (using our `TwoArmHandover` task), but it was using [THIS REPO](https://github.com/ikostrikov/pytorch-a2c-ppo-acktr-gail) instead of stable-baselines-3.

danielstankw, 2022-03-24:
I tried to run 10 episodes for environment experience collection in two ways:
1) 1 process = 1 env with 10 consecutive envs,
2) 5 processes = 5 env in parallel using `SubVecProcEnv` with 2 consecutive episodes
I measured time it took to run those (just env run time, not learning) and compared,

They are almost identical i.e. 1,15sec vs 1.20 sec, which might be due to different seed of simulation i.e. negligible difference.
Any clue why that might be the case @cremebrule

cremebrule, 2022-08-09:
Hmmmm, that's a good question. @roberto-martinmartin do you remember what setup you had when you had used sb3 PPO before, e.g.: in your VICES work?

roberto-martinmartin, 2022-08-24:
Hi!
I did gain with the parallelization. If I remember correctly, I used 10 parallel envs to collect experiences.
Could you check how many times each env is running, just with some simple prints on the subvecprocenv? additionally, maybe there is some very large overhead time that is dominating everything, could you check that? As if creating the env takes 1.10 secs, and the actual episodes are just in the ms, then, for so few episodes it wouldn't be a noticeable difference between parallel or series.
I hope this helps! And sorry for the delay

danielstankw, 2022-08-25:
I tested it on different PC and it did improve results significantly.
Therefore for the parallel learning to work PC one has to use fast CPU one with multiple threads.

closed 2022-08-25 by danielstankw