ARISE-Initiative/robosuite #299: New robot integration error

ashwin-97, 2022-02-24:
I imported my own robot into the package, but when I run the demos, my robot is not found in the selection menu that is in robosuite. I followed the rudimentary guide and running the demo_random_action.py. 
My robot xml file is -

```
<mujoco model="myrobot">
    <compiler 
    angle="radian" 
    meshdir="/usr0/home/amisra/catkin_ws/src/meshes-mujoco/"
    balanceinertia="true"
    discardvisual="false"/>
    <size njmax="500" nconmax="100" />
    <asset>
        <mesh name="base" file="base.stl" />
        <mesh name="shoulder" file="shoulder.stl" />
        <mesh name="upperarm" file="upperarm.stl" />
        <mesh name="forearm" file="forearm.stl" />
        <mesh name="wrist1" file="wrist1.stl" />
        <mesh name="wrist2" file="wrist2.stl" />
        <mesh name="wrist3" file="wrist3.stl" />
        <mesh name="robotiq_85_coupler" file="robotiq_85_coupler.stl" />
        <mesh name="robotiq_85_base_link" file="robotiq_85_base_link.stl" />
        <mesh name="robotiq_85_knuckle_link" file="robotiq_85_knuckle_link.stl" />
        <mesh name="robotiq_85_finger_link" file="robotiq_85_finger_link.stl" />
        <mesh name="robotiq_85_inner_knuckle_link" file="robotiq_85_inner_knuckle_link.stl" />
        <mesh name="robotiq_85_finger_tip_link" file="robotiq_85_finger_tip_link.stl" />
    </asset>
    <worldbody>
        <geom size="0.6 1.25 0.015" pos="0 0 0.985" type="box" />
        <geom size="0.02 0.5" pos="0.55 1.2 0.47" type="cylinder" />
        <geom size="0.02 0.5" pos="-0.55 1.2 0.47" type="cylinder" />
        <geom size="0.02 0.5" pos="0.55 -1.2 0.47" type="cylinder" />
        <geom size="0.02 0.5" pos="-0.55 -1.2 0.47" type="cylinder" />
        <geom size="0.05 1.26 0.025" pos="0.1 0 1.03" type="box" rgba="0.7 0.7 0.7 1" />
        <body name="robot1_base_link" pos="0.1 -1.2 1.05">
            <inertial pos="0 10 0" quat="0.5 0.5 -0.5 0.5" mass="4" diaginertia="0.0072 0.00443333 0.00443333" />
            <joint name="robot1_sliding_base_joint" pos="0 0 0" axis="0 1 0" type="slide" limited="true" range="0 2.4" />
            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="base" />
            <body name="robot1_shoulder_link" pos="0 0 0.089159">
                <inertial pos="0 0 0" mass="3.7" diaginertia="0.0102675 0.0102675 0.00666" />
                <joint name="robot1_shoulder_pan_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" />
                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="shoulder" />
                <body name="robot1_upper_arm_link" pos="0 0.13585 0" quat="0.707107 0 0.707107 0">
                    <inertial pos="0 0 0.28" mass="8.393" diaginertia="0.226891 0.226891 0.0151074" />
                    <joint name="robot1_shoulder_lift_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" />
                    <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="upperarm" />
                    <body name="robot1_forearm_link" pos="0 -0.1197 0.425">
                        <inertial pos="0 0 0.196125" mass="2.275" diaginertia="0.0312168 0.0312168 0.004095" />
                        <joint name="robot1_elbow_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-3.14159 3.14159" />
                        <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="forearm" />
                        <body name="robot1_wrist_1_link" pos="0 0 0.39225" quat="0.707107 0 0.707107 0">
                            <inertial pos="0 0.093 0" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                            <joint name="robot1_wrist_1_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" />
                            <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist1" />
                            <body name="robot1_wrist_2_link" pos="0 0.093 0">
                                <inertial pos="0 0 0.09465" mass="1.219" diaginertia="0.0025599 0.0025599 0.0021942" />
                                <joint name="robot1_wrist_2_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319" />
                                <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist2" />
                                <body name="robot1_wrist_3_link" pos="0 0 0.09465">
                                    <inertial pos="0.000245838 0.0846576 3.50772e-06" quat="0.499851 0.500159 -0.499861 0.500129" mass="0.992851" diaginertia="0.00141553 0.00136577 0.000654231" />
                                    <joint name="robot1_wrist_3_joint" pos="0 0 0" axis="0 1 0" limited="true" range="-6.28319 6.28319" />
                                    <geom type="mesh" rgba="0.7 0.7 0.7 1" mesh="wrist3" />
                                    <geom pos="0 0.0863 0" quat="0.5 -0.5 -0.5 -0.5" type="mesh" rgba="0.7 0.7 0.7 1" mesh="robotiq_85_coupler" />
                                    <geom pos="0 0.0903 0" quat="0.707107 0 0 0.707107" type="mesh" mesh="robotiq_85_base_link" />
                                    <body name="robot1_robotiq_85_left_knuckle_link" pos="-0.0306011 0.145205 0" quat="0 0.707107 0.707107 0">
                                        <inertial pos="-0.00243606 -0.0187741 0" quat="0.647436 0.284299 -0.647436 0.284299" mass="0.0458" diaginertia="4.11145e-05 2.69913e-05 1.81232e-05" />
                                        <joint name="robot1_robotiq_85_left_knuckle_joint" pos="0 0 0" axis="0 0 1" limited="true" range="0 0.80285" />
                                        <geom type="mesh" mesh="robotiq_85_knuckle_link" />
                                        <geom pos="-0.00408552 -0.031486 0" type="mesh" mesh="robotiq_85_finger_link" />
                                    </body>
                                    <body name="robot1_robotiq_85_right_knuckle_link" pos="0.0306011 0.145205 0" quat="0.707107 0 0 0.707107">
                                        <inertial pos="-0.00243606 -0.0187741 0" quat="0.647436 0.284299 -0.647436 0.284299" mass="0.0458" diaginertia="4.11145e-05 2.69913e-05 1.81232e-05" />
                                        <joint name="robot1_robotiq_85_right_knuckle_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-3.14 3.14" />
                                        <geom type="mesh" mesh="robotiq_85_knuckle_link" />
                                        <geom pos="-0.00408552 -0.031486 0" type="mesh" mesh="robotiq_85_finger_link" />
                                    </body>
                                    <body name="robot1_robotiq_85_left_inner_knuckle_link" pos="-0.0127 0.15172 0" quat="0 0.707107 0.707107 0">
                                        <inertial pos="0 0 0" quat="0.707107 0.707107 0 0" mass="0.029951" diaginertia="3.9e-05 3.5e-05 5e-06" />
                                        <joint name="robot1_robotiq_85_left_inner_knuckle_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-3.14 3.14" />
                                        <geom type="mesh" mesh="robotiq_85_inner_knuckle_link" />
                                        <body name="robot1_robotiq_85_left_finger_tip_link" pos="0.0430396 -0.0375994 0">
                                            <inertial pos="0 0 0" quat="0 0.707107 0 0.707107" mass="0.019555" diaginertia="6e-06 5e-06 2e-06" />
                                            <joint name="robot1_robotiq_85_left_finger_tip_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-3.14 3.14" />
                                            <geom type="mesh" mesh="robotiq_85_finger_tip_link" />
                                        </body>
                                    </body>
                                    <body name="robot1_robotiq_85_right_inner_knuckle_link" pos="0.0127 0.15172 0" quat="0.707107 0 0 0.707107">
                                        <inertial pos="0 0 0" quat="0.707107 0.707107 0 0" mass="0.029951" diaginertia="3.9e-05 3.5e-05 5e-06" />
                                        <joint name="robot1_robotiq_85_right_inner_knuckle_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-3.14 3.14" />
                                        <geom type="mesh" mesh="robotiq_85_inner_knuckle_link" />
                                        <body name="robot1_robotiq_85_right_finger_tip_link" pos="0.0430396 -0.0375994 0">
                                            <inertial pos="0 0 0" quat="0 0.707107 0 0.707107" mass="0.019555" diaginertia="6e-06 5e-06 2e-06" />
                                            <joint name="robot1_robotiq_85_right_finger_tip_joint" pos="0 0 0" axis="0 0 1" limited="true" range="-3.14 3.14" />
                                            <geom type="mesh" mesh="robotiq_85_finger_tip_link" />
                                        </body>
                                    </body>
                                </body>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </worldbody>
</mujoco>
```

zhuyifengzju, 2022-03-02:
Hi,

     I would be happy to help with you on this issue. Could you provide more details on how you load this XML file, and what is the command line you get?

ashwin-97, 2022-03-02:
Hello, I created a new folder and followed the rudimentary guide. I have
attached the xml file and I subclass the manipulator model in robots/ and
also import it through __init__

ashwin-97, 2022-03-02:
https://github.com/ashwin-97/Robosuite-Modified

I have uploaded myrobots models/assets/robots folder and models/robots
folder here

zhuyifengzju, 2022-03-13:
Hi, I've looked through your code. I don't think in your current implementation, you have registered your own robot model in it. First of all, you would need to create a python class for your own robot model following something like this: [https://github.com/ARISE-Initiative/robosuite/blob/eef80e5da80ff5a19e617197281336857b5d2db3/robosuite/models/robots/manipulators/panda_robot.py#L7](url), then use the `register_robot` [https://github.com/ARISE-Initiative/robosuite/blob/eef80e5da80ff5a19e617197281336857b5d2db3/robosuite/models/robots/robot_model.py#L10](url) function to register your own class

closed 2022-05-11 by yukezhu