ARISE-Initiative/robosuite #303: XML Error in playback_demonstrations_from_hdf5.py

PraveenElango, 2022-03-13:
Hi, I tried running the following line of code in my robosuite directory to load and play a sample demonstration:

`python playback_demonstrations_from_hdf5.py --folder ../models/assets/demonstrations/wipe/panda`

However, I got the following error:
`Creating window glfw
Playing back random episode... (press ESC to quit)
model_1.xml
Creating window glfw
Traceback (most recent call last):
  File "/home/praveen/robosuite/robosuite/scripts/playback_demonstrations_from_hdf5.py", line 70, in <module>
    xml = postprocess_model_xml(model_xml)
  File "/home/praveen/robosuite/robosuite/utils/mjcf_utils.py", line 511, in postprocess_model_xml
    tree = ET.fromstring(xml_str)
  File "/home/praveen/miniconda3/lib/python3.9/xml/etree/ElementTree.py", line 1348, in XML
    return parser.close()
xml.etree.ElementTree.ParseError: syntax error: line 1, column 0
`
I then changed `tree = ET.fromstring(xml_str)` to `tree = ET.parse(xml_str)`.

However, I now get a new error:

`Creating window glfw
Playing back random episode... (press ESC to quit)
model_1.xml
Creating window glfw
Traceback (most recent call last):
  File "/home/praveen/robosuite/robosuite/scripts/playback_demonstrations_from_hdf5.py", line 70, in <module>
    xml = postprocess_model_xml(model_xml)
  File "/home/praveen/robosuite/robosuite/utils/mjcf_utils.py", line 512, in postprocess_model_xml
    tree = ET.parse(xml_str)
  File "/home/praveen/miniconda3/lib/python3.9/xml/etree/ElementTree.py", line 1229, in parse
    tree.parse(source, parser)
  File "/home/praveen/miniconda3/lib/python3.9/xml/etree/ElementTree.py", line 569, in parse
    source = open(source, "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'model_1.xml'
`
May I ask how can this be fixed? Thank you!

yukezhu, 2022-03-13:
Hi @cremebrule this issue seems to come from the default wipe demonstrations collected on an older version of robosuite. I think the easiest way would be to recollect and update the demo files.

cremebrule, 2022-08-09:
Hmm, I no longer have the spacemouse currently to re-collect demos. @amandlek might you still have a spacemouse by any chance?

yukezhu, 2022-12-04:
The `playback_demonstrations_from_hdf5.py` script should work now with the v1.4 release. Closing this issue now.

closed 2022-12-04 by yukezhu