ARISE-Initiative/robosuite #311: Offscreen rendering - collecting images during human demonstrations

PraveenElango, 2022-03-23:
Hello,

I found out that on-screen rendering is not possible in robosuite when `use_camera_obs` is set to True. Is it possible for me to collect human demonstrations through the visualiser while also collecting image data? I looked at the robomimic framework and figured that it must be possible, since human demonstrations of tasks using MuJoCo and robosuite were collected with image observations. Any help would be appreciated, thank you!

amandlek, 2022-03-23:
The best way to do this is to collect a demonstration file using the on-screen renderer as our collect human demonstrations script does by default, and then follow the guide [here](https://arise-initiative.github.io/robomimic-web/docs/introduction/datasets.html#robosuite-hdf5-datasets) to postprocess the resulting file, and extract image observations from the raw mujoco states.

closed 2022-03-23 by amandlek

PraveenElango, 2022-03-24:
Alright, I'll try this out. Thank you!

yquantao, 2022-09-30:
> The best way to do this is to collect a demonstration file using the on-screen renderer as our collect human demonstrations script does by default, and then follow the guide [...]

Could you reshare the link above? It is unvalid now.