ARISE-Initiative/robosuite #312: Saving only successful demonstrations

PraveenElango, 2022-03-27:
Hello,

I was using the `collect_human_demonstrations.py` script to collect demonstrations for the Wipe task in robosuite. When playing back the dataset of my collected demonstrations, I noticed that even demonstrations that were unsuccessful in completing the task (i.e. not all the markers on the table have been wiped) were still added to the dataset. May I ask if there is an option for me to avoid adding unsuccessful demonstrations to my dataset during the collection of demonstrations? Thank you!

amandlek, 2022-03-27:
Only saving successful demonstrations should already be the default (see [this line](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/scripts/collect_human_demonstrations.py#L72)). It would be helpful if you could provide some more details on how we can reproduce this issue on our end. You can also share your demonstration hdf5 here, and the playback command you used.

PraveenElango, 2022-03-27:
Thanks for getting back to me!

When I am in the midst of collecting a demonstration and I press the reset simulation button (I use the SpaceMouse device), even that demonstration which is not complete is added to the dataset. I used the `playback_demonstrations_from_hdf5.py` file to play back the demonstrations from the hdf5 file. I have shared my hdf5 file here, but I am not sure if it will work on your end as I made some changes to the environment.

amandlek, 2022-03-27:
This is a great catch! You're right - by default all demonstrations are saved, including the ones that are unsuccessful. The easiest way to fix this would be to modify the data collection wrapper to keep track of whether each trajectory is successful (this could be checked in the `env.step` call) and filter the unsuccessful ones out when gathering them into a single hdf5 (in [this function](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/scripts/collect_human_demonstrations.py#L84)). We may also incorporate this functionality on our end, but it may take us some time due to time limitations. If you are able to implement this, we'd also be happy to accept a PR for this functionality!

PraveenElango, 2022-03-28:
Sure, I'll try to create a PR for this soon!

yukezhu, 2022-08-01:
Dear @PraveenElango, I am following up on this issue. Would you like to create a PR for this functionality?

PraveenElango, 2022-08-02:
Hi @yukezhu , sorry for the late reply. Yes, I can create a PR for this today.

closed 2022-12-04 by yukezhu