ARISE-Initiative/robosuite #313: Consider Using dm_control for mujoco bindings

lapp0, 2022-03-29:
Currently robosuite requires a fork of `mujoco-py` called `free-mujoco-py`. This is a fork of https://github.com/openai/mujoco-py/ which is in "maintenance mode", has 23 open pull requests the last of which was merged in November 2021.

The [mujoco repo](https://github.com/deepmind/mujoco) and project is run by DeepMind, as is [dm_control](https://github.com/deepmind/dm_control/tree/master/dm_control/mujoco). The dm_control python bindings for mujoco are actively maintained.

yukezhu, 2022-04-03:
dm_control's interfaces have been migrated to the newly released Python binding of [mujoco](https://mujoco.readthedocs.io/en/latest/python.html). We have been having internal discussions on migrating to this new binding, which as you suggested has longer-term support. The migration would take a substantial amount of effort as the new binding offers a lower level of abstraction than mujoco-py. So I anticipate that it will eventually happen but might take several months of effort.

closed 2022-04-18 by yukezhu

jgsimard, 2022-08-23:
Hi, any news on this?

yukezhu, 2022-08-23:
Still working on this. Things have been slower due to the summer hiatus. Should expect this to be out in 1 month.

jgsimard, 2022-08-23:
oki thanks!

beneisner, 2022-08-26:
+1 looking forward to seeing this! the new mujoco bindings might also allow for better, multithread/process support...