ARISE-Initiative/robosuite #316: [iGibson renderer] Is robosuite supposed to support moving cameras?

kevin-thankyou-lin, 2022-04-18:
Hi, thanks for the great code base!

I'm using an `eye_in_hand` camera to collect observations. However, I've noticed that if I don't manually call `_switch_camera(cam_name)` after each step, the renderer / camera's position never changes. Was that supposed to be the case?

If not, the fix is straightforward - removing the if condition in [`camera_rgb()`](https://github.com/ARISE-Initiative/robosuite/blob/5092e805157ca7a255d3692f34a4d51599e72699/robosuite/renderers/igibson/igibson_renderer.py#L284) calls `_switch_camera` and updates the position of the camera:

```
@sensor(modality=modality)
def camera_rgb(obs_cache):

    # Switch to correct camera
    if self is not None and self.camera_name != cam_name: # remove this line
        self._switch_camera(cam_name)
```

Code to reproduce camera position not changing (slight modification of `robosuite/renderers/igibson/igibson_renderer.py`:

```
if __name__ == "__main__":
    config = load_renderer_config("igibson")
    # config['optimized'] = True

    config['vision_modalities'] = ['rgb', 'normal', '3d']
    # config['vision_modalities'] = ['seg']
    config['camera_obs'] = True
    # config['render_mode'] = 'headless'
    # config['msaa'] = False

    env = suite.make(
        "Door",
        robots=["Sawyer"],
        reward_shaping=True,
        has_renderer=False,  # no on-screen renderer
        has_offscreen_renderer=True,  # no off-screen renderer
        ignore_done=True,
        use_object_obs=True,  # use object-centric feature
        use_camera_obs=False,
        render_camera="robot0_eye_in_hand",
        control_freq=10,
        renderer="igibson",
        camera_segmentations="element",
        renderer_config=config,
        camera_names=["robot0_eye_in_hand"],
    )

    obs = env.reset()
    for i in range(10000):
        action = np.random.randn(8)
        obs, reward, done, _ = env.step(action)
        import cv2
        cv2.imwrite(f"{i}.png", (obs['robot0_eye_in_hand_image'][..., :3] * 255).astype(np.uint8)) # to visualize the observations
        env.render()
    env.close()

    print("Done.")
```

roberto-martinmartin, 2022-04-25:
Hi @kevin-thankyou-lin ,
thank you so much for reporting this issue!

We just merged a fix into master. could you check that it solves your case?

Thanks
--Roberto

closed 2022-04-25 by roberto-martinmartin

kevin-thankyou-lin, 2022-04-26:
Thanks so much, Roberto! Things are fixed :)

Kevin

FanZhang91, 2025-04-28:
robosuite==1.4.0
I meet the same problem. I use renderer ="mujoco", how to fix this problem?