ARISE-Initiative/robosuite #321: "No "site" with name <>" error when accessing "_eef_xmat" properties

misoshiruseijin, 2022-05-03:
Accessing `_eef_xmat` property in `single_arm_env`  or the `_eef0_xmat` and `_eef1_xmat` properties in `two_arm_env`, causes an error "Value Error: No "site" with name robot0_ee exists." The error is resolved with these changes to `_eef_xmat`, `_eef0_xmat`, and `_eef1_xmat`:

- replacing "ee" with "grip_site" and "right_ee" with "right_grip_site"
- replacing `pf = self.robots[0].robot_model.naming_prefix` with `pf = self.robots[0].gripper.naming_prefix`

cremebrule, 2022-05-11:
Hi @misoshiruseijin ,

Thanks for finding this bug! Can you submit a PR with the appropriate fixes? Thanks!

cremebrule, 2022-08-09:
Thanks for fixing this! Closing now.

closed 2022-08-09 by cremebrule