ARISE-Initiative/robosuite #322: Hiding coordinate system

danielstankw, 2022-05-03:
Hi,
When I visualise the simulation and press E I am able to see reference frames. 
How can I disable certain frames from being visible? 
I want to see coordinate frames but I am not interested of seeing them for all the joints. 

Also they are massive. Is there a place I can change their size? 

[image]

danielstankw, 2022-05-03:
@cremebrule @roberto-martinmartin

danielstankw, 2022-05-10:
Anyone?

cremebrule, 2022-05-11:
Hi @danielstankw ,

(Reposting the response from slack): I think this is a native mujoco visualization functionality that we don't have control over, IIRC.

You might have to look into the low-level C++ code in order to directly modify how the mujoco renderer handles visualizing the frames.

Closing this issue for now, feel free to reopen or start a new issue if you continue to run into issues.

closed 2022-05-11 by cremebrule