ARISE-Initiative/robosuite #323: Failed to load XML from string. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'

tinelion, 2022-05-07:
I use robosuite on win10 and got error below,I'm sure the assert files exist，what's wrong and what can i do? robosuite==1.3.2
the script from document:
```
import numpy as np
import robosuite as suite

# create environment instance
env = suite.make(
    env_name="Lift", # try with other tasks like "Stack" and "Door"
    robots="Jaco",  # try with other robots like "Sawyer" and "Jaco"
    has_renderer=True,
    has_offscreen_renderer=False,
    use_camera_obs=False,
)

# reset the environment
env.reset()

for i in range(1000):
    action = np.random.randn(env.robots[0].dof) # sample random action
    obs, reward, done, info = env.step(action)  # take action in the environment
    env.render()  # render on display
```

the error:

D:\app\Anaconda\envs\broken\python.exe E:/projects/gym_learn/class_6/run.py
Traceback (most recent call last):
  File "E:/projects/gym_learn/class_6/run.py", line 10, in <module>
    use_camera_obs=False,
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\base.py", line 38, in make
    return REGISTERED_ENVS[env_name](*args, **kwargs)
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\manipulation\lift.py", line 207, in __init__
    renderer_config=renderer_config,
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\manipulation\manipulation_env.py", line 186, in __init__
    renderer_config=renderer_config,
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\robot_env.py", line 226, in __init__
    renderer_config=renderer_config,
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\base.py", line 137, in __init__
    self._initialize_sim()
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\environments\base.py", line 236, in _initialize_sim
    self.mjpy_model = load_model_from_xml(xml_string) if xml_string else self.model.get_model(mode="mujoco_py")
  File "C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\base.py", line 145, in get_model
    model = load_model_from_xml(string.getvalue())
  File "cymj.pyx", line 191, in mujoco_py.cymj.load_model_from_xml
Exception: <mujoco model="base">
  <compiler angle="radian" inertiagrouprange="0 0" meshdir="meshes/" />
  <option cone="elliptic" density="1.2" impratio="20" timestep="0.002" viscosity="0.00002" />
  <size nconmax="5000" njmax="5000" />

  <asset>
  <texture builtin="gradient" height="256" rgb1=".9 .9 1." rgb2=".2 .3 .4" type="skybox" width="256" />
    <texture file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\arenas\../textures/wood-tiles.png" name="texplane" type="2d" />
    <material name="floorplane" reflectance="0.01" shininess="0.0" specular="0.0" texrepeat="2 2" texture="texplane" texuniform="true" />
    
    <texture file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\arenas\../textures/ceramic.png" name="tex-ceramic" type="cube" />
    <material name="table_ceramic" reflectance="0.0" shininess="0.0" specular="0.2" texrepeat="1 1" texture="tex-ceramic" />
    
    <texture file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\arenas\../textures/steel-brushed.png" name="tex-steel-brushed" type="cube" />
    <material name="table_legs_metal" reflectance="0.8" shininess="0.8" texrepeat="1 1" texture="tex-steel-brushed" />
    
    <texture file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\arenas\../textures/cream-plaster.png" name="tex-cream-plaster" type="2d" />
    <material name="walls_mat" reflectance="0.0" shininess="0.1" specular="0.1" texrepeat="3 3" texture="tex-cream-plaster" texuniform="true" />
    
    <texture builtin="flat" height="512" name="textable" rgb1="0.5 0.5 0.5" rgb2="0.5 0.5 0.5" width="512" />
    <material name="table_mat" texture="textable" />

  <mesh file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\robots\jaco\meshes/base.stl" name="robot0_base" />
        <mesh file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\robots\jaco\meshes/shoulder.stl" name="robot0_shoulder" />
        <mesh file="C:\Users\xxxxxx\AppData\Roaming\Python\Python37\site-packages\robosuite\models\assets\robots\jaco\meshes/ring_big.stl" name="robot0_ring_big" />
[... 228 lines cut ...]

        <tendon name="gripper0_pinky_12_cpl" tendon1="gripper0_pinky_12_cpl" />
	</equality>
<sensor><force name="gripper0_force_ee" site="gripper0_ft_frame" />
        <torque name="gripper0_torque_ee" site="gripper0_ft_frame" />
    </sensor><tendon><fixed frictionloss="0.4" name="gripper0_thumb_cpl" range="-5 5" springlength="0.2" stiffness="3.0">
			<joint coef="0.4" joint="gripper0_joint_thumb" />
			<joint coef="-0.4" joint="gripper0_joint_thumb_distal" />
		</fixed>

		
        <fixed frictionloss="0.4" name="gripper0_index_12_cpl" range="-5 5" springlength="0.2" stiffness="3.0">
			<joint coef="0.4" joint="gripper0_joint_index" />
			<joint coef="-0.4" joint="gripper0_joint_index_distal" />
		</fixed>

		
        <fixed frictionloss="0.4" name="gripper0_pinky_12_cpl" range="-5 5" springlength="0.2" stiffness="3.0">
			<joint coef="0.4" joint="gripper0_joint_pinky" />
			<joint coef="-0.4" joint="gripper0_joint_pinky_distal" />
		</fixed>
	</tendon><contact /></mujoco>
Failed to load XML from string. mj_loadXML error: b'XML parse error at line 0, column 0:\nFailed to open file\n'

Process finished with exit code 1

ldepn, 2022-09-02:
I meet the same issue, did you have solve？

tobi-ostfalia, 2023-04-20:
I'm on robosuite 1.4.0 and I have the same error, and one more:
`ModuleNotFoundError: No module named 'robosuite.macros_private'`
`FileNotFoundError: [Errno 2] No such file or directory: 'C:\\tmp\\robosuite.log'`
Both are on line 0, at the import.

ldepn, 2023-04-20:
您的邮件已收到，我会尽快处理！

junaun, 2023-09-06:
> I'm on robosuite 1.4.0 and I have the same error, and one more: `ModuleNotFoundError: No module named 'robosuite.macros_private'` `FileNotFoundError: [Errno 2] No such file or directory: 'C:\\tmp\\robosuite.log'` Both are on line 0, at the import.

I facing the same issue when installing robosuite 1.4.0 on Windows 10 with conda env python3.8, this is because robosuite only support Linux and macOS. I have found a solution from this article written in Chinese: https://zhuanlan.zhihu.com/p/607578201

1.  For this error: `FileNotFoundError: [Errno 2] No such file or directory: 'C:\\tmp\\robosuite.log'`, create a folder named 'tmp' under C:\
2. Next you will face `mujoco.dll not found` error, go to `anaconda3\envs\{your env name}\Lib\site-packages\mujoco`, copy mujoco.dll and paste it to `anaconda3\envs\{your env name}\Lib\site-packages\robosuite\utils`
3. Go to binding_utils.py inside the robosuite\utils folder, go to line 43 and change `egl` in  `os.environ["MUJOCO_GL"] = "egl"` to `"wgl"`
```
if macros.MUJOCO_GPU_RENDERING and os.environ.get("MUJOCO_GL", None) not in ["osmesa", "glx"]:
    # If gpu rendering is specified in macros, then we enforce gpu
    # option for rendering
    if _SYSTEM == "Darwin":
        os.environ["MUJOCO_GL"] = "cgl"
    else:
        os.environ["MUJOCO_GL"] = "wgl"
```
4. Should be working now, run the test: `python -m robosuite.demos.demo_random_action`

LukeLIN-web, 2025-05-23:
> ModuleNotFoundError: No module named 'robosuite.macros_private'

Same problem in 1.5.1 lastest