ARISE-Initiative/robosuite #324: [Question] Is custom task possible?

ziji-wu, 2022-05-10:
Hi! In my project, I want to use the UR5 robot with a new task that seems like the registered task "Pick-and-Place" a little bit. So I wonder is there any convenient API for users to construct their own task and do simulation?

From the doc, when making the envs by `suite. make()`, the registered task will be instantiated in a private method `_load_model`, is that mean one must change here to register their own task? If so, that seems too time-consuming and pretty easy to make mistakes for new users compared with developers involved in this repo.

Could you please give me some guidelines if possible? Thanks!

Dhanushvarma, 2023-11-21:
Does robosuite now support custom tasks?

yukezhu, 2023-12-29:
Yes, the best way to create your own tasks is to take a look at the examples in the [environments](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/environments) folder and implement the new tasks in the same way.

closed 2023-12-29 by yukezhu