ARISE-Initiative/robosuite #328: How to add shapenet object

huisyy, 2022-05-19:
I'm trying to add a shapenet object. I understand the structure of objects, but I'm really struggling to create the xml/png texture files, as specified in [robosuite](https://robosuite.ai/docs/modules/objects.html)--right now, I'm trying to use Blender to create an object, and then export into the xml/png files respectively. I was wondering how you guys created the native objects in robosuite, and if there's a better way to do so than what I'm currently doing? Thanks!

snasiriany, 2022-08-01:
Hi,

Are you looking to simulate shape shapes (boxes, cylinders, etc) or more complex shapes (a shoe, mug, etc)? For the former case robosuite offers a number of [primitive shapes](https://github.com/ARISE-Initiative/robosuite/tree/master/robosuite/models/objects/primitive) that you can programmatically create, without creating any xml manually (see [here](https://github.com/ARISE-Initiative/robosuite/blob/6a23c470c55b062fc0e493609eab152ddbf97f4e/robosuite/environments/manipulation/cleanup.py#L485) for an example).

However if you are interested in more complex shapes, the process is more involved:
1. First, we recommend switching to the latest mujoco version to support `.obj` object meshes. This feature was introduced in mujoco [2.1.2](https://github.com/deepmind/mujoco/releases/tag/2.1.2), and you would need to use a compatible `mujoco-py` version that supports this version of mujoco.
2. You need to generate a convex collision mesh. This can be done with the `trimesh` collision package, or [V-HACD](https://github.com/kmammou/v-hacd).
3. Then you need to create a new xml for the object, using the textures and meshes you created in the previous steps. You can perhaps follow the example [here](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/objects/can.xml) for reference.

Just to let you know, our team is working on a pipeline to automatically port ShapeNet objects into robosuite. This is still in the works, we plan to release these features to the public in the coming months. Please let us know if you have additional questions.

Soroush

Harimus, 2022-09-16:
Hello I have a question: for step 1, I can't find a Mujoco 2.1.2 compatible mujoco-py (the official one supports `mujoco210`) is there some fork that make this work, or is it possible to run robosuite on >2.1.2 mujoco which is possible to install with `pip install mujoco`?

snasiriany, 2022-09-17:
Thanks for the question. Indeed it is difficult to rely on `mujoco-py` for later versions of Mujoco. We are working right now on removing the dependency on `pip install mujoco-py` and instead using `pip install mujoco`. This involves a number of changes under the hood. We are planning to release this version in the coming weeks.

Harimus, 2022-09-21:
Hi, thank you for the reply! Ok I guess I'll wait for now, but is there a known workaround to this if I want to move on with this while waiting?

otherwise, do you need help/take pr related to this dependency replacement?

snasiriany, 2022-09-21:
I tried these steps a while back for Ubuntu, hopefully they end up working for you:

* Download `mujoco-2.1.1` into `~/.mujoco` directory
* Make a “fake” `mujoco210` folder in the `~/.mujoco` directory, and make an empty `bin` folder in `mujoco210`

* Add 4 symlinks exactly like this:

```
ln -sf ~/.mujoco/mujoco-2.1.1/lib/libglewegl.so ~/.mujoco/mujoco210/bin/libglewegl.so
ln -sf ~/.mujoco/mujoco-2.1.1/lib/libmujoco.so ~/.mujoco/mujoco210/bin/libmujoco210.so

ln -sf ~/.mujoco/mujoco-2.1.1/lib/libglewegl.so ~/.mujoco/mujoco-2.1.1/bin/libglewegl.so
ln -sf ~/.mujoco/mujoco-2.1.1/lib/libmujoco.so ~/.mujoco/mujoco-2.1.1/bin/libmujoco210.so
```

* Make sure your ~/.bash_rc contains these lines:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.mujoco/mujoco210/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/.mujoco/mujoco-2.1.1/bin
export MUJOCO_PY_MUJOCO_PATH=~/.mujoco/mujoco-2.1.1
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/lib/nvidia
```

closed 2022-12-04 by yukezhu

kevin-thankyou-lin, 2023-08-05:
Noting here for reference (needed for older versions of robosuite):

1. Export the object from blender as .obj: this gives both the .obj and the .mtl files
2. Export the object as .stl
3. Use [obj2mjcf](https://github.com/kevinzakka/obj2mjcf) to get the correct .xml file structure
4. Replace the .obj with the .stl (required for older versions of robosuite; not required for newer versions at the time of this post)