ARISE-Initiative/robosuite #330: Mass matrix different than expected value

shreysahgal, 2022-05-20:
Hello,

I'm trying to verify the dynamics of a Fetch robot which I've imported into Robosuite. I'm doing this by generating random configurations, getting the mass matrix of the robot at each of the configurations, and comparing with mass matrices generated in Matlab simulation. It seems like no matter how I tune the inertial parameters in the robot.xml file, the mass matrix seems to be on the incorrect order of magnitude -- the Matlab simulation generally gives numbers on the order of 10^-2, but in Robosuite I get numbers on the order of 10^1.

I ran the same test using the Kinova3 arm to see if there was some problem with the way I imported the robot into Robosuite, but after generating configurations, getting the mass matrices in Robosuite, and getting baseline mass matrices in Matlab, I get the same behavior (i.e. the mass matrix seems to be wrong).

The way I get the mass matrix in Robosuite is:
```
env = suite_traj.suite.make(
    env_name = "Empty",
    robots = "Kinova3",
    has_renderer=False,
    has_offscreen_renderer=False,
    use_camera_obs=False,
    controller_configs = controller_config,
    control_freq=2,
    render_collision_mesh=True
)

q = [ ... ] # some configuration
Kinova3.init_qpos = q
mass_matrix = env.robots[0].controller.mass_matrix
```
I found that using `set_robot_joint_positions` didn't seem to change the mass matrix (which I think makes sense since the mass matrix is retrieved through the controller), so I manually set the `init_qpos`.

Is there something wrong with my method, or is there something awry with the mass matrix of the Kinova robot/the Fetch robot I imported? Thanks!

shreysahgal, 2022-05-20:
After some more testing, I found that while the mass matrix given by the Robosuite controller API was incorrect, the mass matrix returned by the underlying `mujoco_py.functions.mj_fullM` function was correct (compared to the baseline Matlab simulation). So I figure either I'm retrieving the mass matrix incorrectly, or there is some problem in the code which changes the mass matrix from the underlying function call. This problem persists with built-in Robosuite models too, not just imported robots.

closed 2022-12-04 by yukezhu

anselmo-parnada, 2025-03-11:
Hi,

This is an old post, but I found myself with the same issue. Instead, I had different mass matrices between Pinocchio and Robosuite.

Luckily, I found a reply on issue [#796](https://github.com/openai/mujoco-py/issues/796#issuecomment-2590930033) in the [mujoco-py](https://github.com/openai/mujoco-py) repository by @luismuschal that helped me debug the issue. I will relay the information here in case it helps someone in the future.

Basically, armature in the joints contribute to the computed mass matrix. Specifically, it manifests as additional inertia in the diagonal elements of the mass matrix. If there are no armature values for the joints set in the loaded `robot.xml` file,  Robosuite automatically assigns armature values to the joints of your robot (see [robot_model.py](https://github.com/ARISE-Initiative/robosuite/blob/57e5d3bda6ed88aab5204297bec96792ed4130d4/robosuite/models/robots/robot_model.py#L78)), so it's quite easy to not account for armature when computing mass matrices outside of Mujoco.

So if you notice that mass matrices between MATLAB and Robosuite are similar apart from the diagonal terms, this may be why!