ARISE-Initiative/robosuite #339: Kinova Gen3 MJCF inconsistent with hardware

rbharadwaj9, 2022-06-23:
Hi,

It looks like the MJCF model in robosuite for the [Kinova Arm](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/assets/robots/kinova3/robot.xml) is inconsistent with the hardware robot. Mainly, the joint axes seem to be flipped for all the joints, essentially causing the robot to be a mirror image of the real robot in the Home position, as specified by the Kinova documentation. 

It looks like the reference frames are correct with respect to the URDF file that I visualized using MATLAB. 
[image]

It may be the case that this is purely a visual thing but I am not sure at the moment and this may lead to inconsistencies for computing dynamics between sim and reality.

melfm, 2022-07-12:
I found similar inconsistency here [issue](https://github.com/ARISE-Initiative/robosuite/issues/263), not for this model but for the Jaco arm. I had to convert the measurements coordinates, but its not very convenient for sure. I think it'd be better to match them at the modelling level.

francelico, 2023-02-24:
@rbharadwaj9 did you manage to sort the issue out in the end?