ARISE-Initiative/robosuite #341: Robosuite with Garage

qureshinomaan, 2022-06-27:
Hi. 
I am trying to run Robosutie Environment with garage(https://garage.readthedocs.io)
I wrote a simple wrapper for robosuite, however it is giving this error : """TypeError: no default __reduce__ due to non-trivial __cinit__""". Seems like their is some error while multi-processing.
Thanks!

danielstankw, 2022-08-28:
Hi,
I cant help you with your question but I recommend using Stable Baselines 3 for implementation of RL for robosuite. 
I have experience in that field so if you decide to use it instead I could help.

Unfortunately this github isn't very active so you can wait months before someone will answer.

kevin-thankyou-lin, 2024-11-07:
Ah, looks like folks never replied to this issue --- for folks trying out RL w/ robosuite, see this [solution](https://github.com/ARISE-Initiative/robosuite/issues/483)

closed 2024-11-07 by kevin-thankyou-lin