ARISE-Initiative/robosuite #345: Can't write stl file path into mujoco xml

hooyahliu, 2022-07-07:
Thank you for your excellent work！
When I follow the tutorial  [Creating a XMLObject](https://robosuite.ai/docs/modules/objects.html#creating-a-xmlobject) to try to create a bread object and load it with `BreadObject(name='bread').get_obj()`,  an error occurred:
`Failed to load XML from string. mj_loadXML error: b"Error: mesh 'bread_mesh' not found in geom 30`.

Can the path of the stl file be added automatically?

hooyahliu, 2022-07-08:
> Hi, Do you mind giving a little bit more context about how you are calling this line? If you do not have trouble initializating pick-place environment:
> 
> https://github.com/ARISE-Initiative/robosuite/blob/361c136c47b93b12381dfdf0463729812d307628/robosuite/environments/manipulation/pick_place.py#L790
> 
> , the issue you are experiencing is probably because of calling the object initialization incorrectly.

thank you for your reply！
I have no trouble  initializating pick-place environment.

I am trying to modify this line of program:
https://github.com/ARISE-Initiative/robosuite/blob/361c136c47b93b12381dfdf0463729812d307628/robosuite/demos/demo_gripper_interaction.py#L45

So I directly changed ` mujoco_object = BoxObject(
        name="box", size=[0.02, 0.02, 0.02], rgba=[1, 0, 0, 1], friction=[1, 0.005, 0.0001]
    ).get_obj()` to `mujoco_object = BreadObject(name='bread').get_obj()`

Is this initialization method wrong?

zhuyifengzju, 2022-07-08:
I see. So when a box object is initialized, it doesn't have a specific mesh file, and it can be directly appended to `worldbody`. In the case of bread, not only it needs to be added to `worldbody`, the assets also need to be added. See [here](https://github.com/ARISE-Initiative/robosuite/blob/361c136c47b93b12381dfdf0463729812d307628/robosuite/models/tasks/task.py#L88). 

In short, a simple solution for you is like this:
`bread_object = BreadObject(name="bread)`
`mujoco_object = bread_object.get_obj()`
`world.merge_assets(bread_object)`
`world.worldbody.append(mujoco_object)`

Let me know if you have any more questions!

hooyahliu, 2022-07-09:
Thank you very much!
This problem has been solved!

But a new problem arose.
I try to set BreadObject's position and Quaternion with:
       ` self.bread = BreadObject(name='bread')`
       ` mujoco_obj = self.bread.get_obj()`
       ` mujoco_obj.set('pos', array_to_string(obj_pos))`
       ` mujoco_obj.set('quat', array_to_string(obj_quat))`
       ` world.merge_assets(self.bread)`
       ` world.worldbody.append(mujoco_obj)`

the `obj_pos`is `"0.1 0.1 0.1"`, and the `obj_quat` is `"0.924 0.3826 0 0"`
But after I save and open this xml file, weired spaces appearing in these numbers:
`<body name="bread_main" pos="0 . 1   0 . 1   0 . 1" quat="0 . 9 2 4   0 . 3 8 2 6   0   0">`
[image]

hooyahliu, 2022-07-10:
Thanks for your help, my problem has been solved!!!
O(∩_∩)O

zhuyifengzju, 2022-07-10:
No problem! I'll close this issue then.

closed 2022-08-01 by yukezhu