ARISE-Initiative/robosuite #347: How to set the random seed for environments?

Altriaex, 2022-07-14:
Hi, I am new to robosuite, and I would like to use it in my experiments. Could you please tell me how to set the random seed for environment to improve reproducibility of experiments?

zhuyifengzju, 2022-07-18:
Hi,
The randomness of environment initializations are mainly based on numpy random generators. And the randomness of policies come from deep learning packages (such as torch). Therefore, in order to set the random seed of robosuite environments,  you can do: `np.random.seed(SEED_VALUE)`, `torch.manual_seed(SEED_VALUE)` (as in robomimic repo: [https://github.com/ARISE-Initiative/robomimic/blob/b5d2aa9902825c6c652e3b08b19446d199b49590/robomimic/scripts/run_trained_agent.py#L216](https://github.com/ARISE-Initiative/robomimic/blob/b5d2aa9902825c6c652e3b08b19446d199b49590/robomimic/scripts/run_trained_agent.py#L216)).

Altriaex, 2022-07-19:
Thank you for your replay.

closed 2022-07-19 by Altriaex