ARISE-Initiative/robosuite #354: How do we access the goal state for a given robosuite task?

ago109, 2022-09-26:
Hello,

One critical question about the functionality offered by currently implemented robosutie environments -- if a user wanted to, at the start of a given episode for a given task, such as "PickPlaceCan", to retrieve the goal state, how would one does so? 
For example, the "object-state" contains the current conditions/state of the object of interest for the robot, but how do we either access (through the current task API or hack) the "object-state" of a specific episode/run of "PickPlaceCan" at the desired goal position (the "object-state" when the robot arm has successfully deposited the can at the next bin to terminate the episode)?

I examined the design of the reward/_check_success() function and I do not, at least at first glance, see where the "object-state" is accessed or measured -- if the who 10-dimensional "object-state" at the goal position is not available, how would we compose the goal "object-state" from the information available within the "PickPlaceCan" environment task?

This would be invaluable for directly testing RL methodologies based on hindsight experience replay (HER) as HER does actually require access to the vector form of a goal state so the agent can receive this as input and move towards (according to universal value function approximation) even if it does throw in some "virtual/fake" sub-goal states during the training process. The old Mujoco robotic arms tasks (like Fetch-Reach-v1, etc.) offered the ability to directly access the goal states which was and is core to HER's actual performance.

Again, if easy direct access to task goal states is not implemented, could you please offer some guidance as to how to compose the goal state from the task environment ourselves? (this would need to be done per episode, as the goal state would be different for each episode variation of the task, e.g., random object initialization, etc.)

datonefaridze, 2022-10-19:
I also need that info, @ago109 did you find that out?

ago109, 2022-10-27:
@datonefaridze no, I never heard back from the robosuite team nor did I figure any real useful solution. Thankfully, I was able to work around my need for a goal state through other means (in other words, not using the goal state). 
Digging into the source, however, the goal state is not technically in one spot -- it's sort of implied by the reward function and the problem environment, so one would have to write some sort of backwards solver to find the true goal state for most of these tasks (at least from my initial impression and dig through of the source code -- I might have missed something along the way and certainly would love the robosutie devs to point me and others in the right direction for extracting the true goal for some of these problems).

This would still be a highly desirable item to have in robosuite, and I feel something like this should be implemented in future iterations of the library.

datonefaridze, 2022-10-28:
@ago109 I raised another issue and they told me that they would take a look a that problem.