ARISE-Initiative/robosuite #355: Python 3.10 support

jdekarske, 2022-09-26:
The current numba dependency does not support Python 3.10:

```
  File "/tmp/easy_install-hl2uylwf/numba-0.53.1/setup.py", line 47, in _guard_py_ver
RuntimeError: Cannot install on Python version 3.10.6; only versions >=3.6,<3.10 are supported.
```

Using `"numba>=0.52.0",` seems to work, but I haven't fully tested.

jhejna, 2023-01-12:
Numba seems to now support python 3.10, so the above issue should be resolved. However, there are a few Typing name that were deprecated in Python 3.10, particularly those that moved from `collections` to `collections.abc`:
```
  File "[hidden]/robosuite/robosuite/models/arenas/multi_table_arena.py", line 6, in <module>
    from collections import Iterable
ImportError: cannot import name 'Iterable' from 'collections' ([hidden]/lib/python3.10/collections/__init__.py)
```
A simple find and replace in the code would likely fix this.

minhnh, 2023-05-18:
I believe you missed one instance of this `Iterable` import issue in [`models/arenas/multi_table_arena.py`](https://github.com/ARISE-Initiative/robosuite/blob/de64fa5935f9f30ce01b36a3ef1a3242060b9cdb/robosuite/models/arenas/multi_table_arena.py#L1).

yukezhu, 2023-05-18:
Thanks @minhnh. Do you mind submitting a PR to get this fixed?

minhnh, 2023-05-19:
@yukezhu I submitted #402 as a fix.

PR #402 (2023-05-19): fix Iterable import for python 3.10 in multi_table_arena
- import `Iterable` from the correct module in `robosuite/models/arenas/multi_table_arena.py`
- resolve #355
closed 2023-05-19 by yukezhu