ARISE-Initiative/robosuite #357: Can i record image observations while collecting data

datonefaridze, 2022-10-23:
Hello is it possible to collect image observations while collecting data? I followed this [documentation](https://robosuite.ai/docs/algorithms/demonstrations.html) , but it doesn't state who to do it, it only collects state observations, and i can't have: ` has_renderer=True, has_offscreen_renderer=True,` at the same time

snasiriany, 2022-10-24:
robomimic allows you to extract image observations from an existing dataset collected in robosuite. Please follow the instructions here: https://robomimic.github.io/docs/datasets/robosuite.html

datonefaridze, 2022-10-24:
@snasiriany Yeah I know it but i just wanted to collect my own data, I mean that data has to be collected somehow, so how they did it? I think it will be nice if you add feature to the codebase which allows you to do that.
Also it would be better if you add goal_state in observation, one day i was trying to use hindsight replay, but couldn't because envrionment doesn't return goal_state, but it is easy to add (for Lift just add hight to beginning position)

snasiriany, 2022-10-24:
The data that is collected contains raw, low-level mujoco states. It would be nice to collect images as well on the spot in the robosuite data collection script, I'll let @amandlek comment here if we should support that.

Regarding goal state - how do we determine the specific goal state for each episode? For several of our environments there are many possible goal states. That's my main concern.

datonefaridze, 2022-10-24:
@snasiriany I mean global goal state, for Lify if it starts randomly goal state would be (x,y,z+h) where h is desired hight. if it is pick and place desired state would be middle point in a bin (if there are many discrete goals i would suggest to output just one). in general i agree about challenge of goal state, but i think it would be nice if you figure that oud. if it's not possible it's okay.

snasiriany, 2022-10-24:
Thanks for the suggestion, I'll run it by our team later this week and you know what we decide

closed 2023-05-01 by yukezhu

Dhanushvarma, 2023-11-12:
For collecting image observation during collecting demonstrations, is the robomimic method still the only way (preprocess and extract using robomimic conversion scripts)? I ask because, I am having trouble installing robomimic.

snasiriany, 2023-11-13:
I believe doing things through robomimic is the only way. I tried responding to your other Issue in robomimic