ARISE-Initiative/robosuite #359: In OSC mode, the action and motion of UR5e robot are not consistent?

dexin-wang, 2022-10-28:
I transmit the action loop to the UR5e, and then observe the returned robot0_eef_pos:
code:
`for i in range(10000):
        action = [0, 0, 0.01, 0, 0, 0, 0]
        obs, reward, done, _ = env.step(action)
        print('robot0_eef_pos = ', obs['robot0_eef_pos'])
        env.render()`
result:
`robot0_eef_pos =  [-0.25421061 -0.00983867  0.96652722]
robot0_eef_pos =  [-0.25423838 -0.00981668  0.96661474]
robot0_eef_pos =  [-0.25424918 -0.00980347  0.96673156]
robot0_eef_pos =  [-0.25426361 -0.00979126  0.9668497 ]
robot0_eef_pos =  [-0.25427933 -0.00978154  0.96696851]
robot0_eef_pos =  [-0.25429542 -0.00977323  0.96708783]
robot0_eef_pos =  [-0.25431136 -0.00976523  0.9672075 ]
robot0_eef_pos =  [-0.2543267  -0.00975754  0.96732764]
robot0_eef_pos =  [-0.25434131 -0.00975022  0.9674483 ]
robot0_eef_pos =  [-0.25435516 -0.00974329  0.96756949]
robot0_eef_pos =  [-0.25436827 -0.00973673  0.9676912 ]
robot0_eef_pos =  [-0.25438063 -0.00973054  0.96781343]
robot0_eef_pos =  [-0.2543923  -0.0097247   0.96793614]
robot0_eef_pos =  [-0.25440337 -0.00971917  0.96805927]
robot0_eef_pos =  [-0.25441403 -0.00971383  0.96818265]
...`

hai-h-nguyen, 2023-05-23:
Probably the OSC controller does not work perfectly; therefore, the end-effector cannot perfectly keep the same position.

ChenWEIx2, 2025-02-26:
have you fixed this problem?