ARISE-Initiative/robosuite #364: Cannot initialize a EGL device display

lorepieri8, 2022-12-01:
To reproduce, follow "install from pip" instructions:
https://robosuite.ai/docs/installation.html

Error:

```
~$ python -m robosuite.demos.demo_random_action
[robosuite WARNING] No private macro file found! (__init__.py:7)
[robosuite WARNING] It is recommended to use a private macro file (__init__.py:8)
[robosuite WARNING] To setup, run: python /home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/scripts/setup_macros.py (__init__.py:9)
Traceback (most recent call last):
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/__init__.py", line 11, in <module>
    from robosuite.environments.base import make
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/environments/__init__.py", line 1, in <module>
    from .base import REGISTERED_ENVS, MujocoEnv
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/environments/base.py", line 12, in <module>
    from robosuite.utils.binding_utils import MjRenderContextOffscreen, MjSim
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/utils/binding_utils.py", line 58, in <module>
    from robosuite.renderers.context.egl_context import EGLGLContext as GLContext
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/robosuite/renderers/context/egl_context.py", line 32, in <module>
    from mujoco.egl import egl_ext as EGL
  File "/home/dummy/miniconda3/envs/robosuite/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 67, in <module>
    raise ImportError(
ImportError: Cannot initialize a EGL device display. This likely means that your EGL driver does not support the PLATFORM_DEVICE extension, which is required for creating a headless rendering context.

```
The suggested solution ("To setup, run: python ...") results in the same error.

lorepieri8, 2022-12-02:
Do you have any graphics requirements? I've Mesa Intel® UHD Graphics (TGL GT1), but could switch to Nvidia if this helps. 

I'm not familiar with EGL, how would I know what I'm running and what extensions I can support?

zhuyifengzju, 2022-12-02:
The graphics requirements depend on the mujoco. There is another gl driver option based on osmesa that you might want to try. For more details of switching gl opptions in robosuite, please refer to: https://github.com/ARISE-Initiative/robosuite/blob/eb6fecc93f881a648b2a4096ec31a44503c7db9b/robosuite/utils/binding_utils.py#L29.

lorepieri8, 2022-12-05:
Solved by installing the proper Nvidia drivers. Make sure nvidia-smi returns your GPU stats.

closed 2022-12-05 by lorepieri8

GuHuangAI, 2024-06-14:
Hello, I have the same problem but there are no useful solutions.

GuHuangAI, 2024-06-14:
> There is another

nvidia-smi can output the GPU stats.