ARISE-Initiative/robosuite #366: Contact detection fails in between control timesteps

jbonyun, 2022-12-08:
We have a task in which contact between two objects is brief (we're hitting a ball). If the contact begins and ends in between control timesteps, then Robosuite will not report any contact has happened when you call `check_contact` or `get_contacts`.

It's not *wrong* exactly -- there is no contact at that instant -- but it is not useful for evaluating whether contact *has occurred*.

We worked around this in our `Environment` by hooking into `_update_observables`, which is called after every Mujoco timestep. In there, we check for contacts we care about, and keep a logical-or'ed running tally of the contact during each control timestep. We reset this tally in `step` and `reset` for the next control timestep.

I think it would be useful if Robosuite was able to perform this automatically. Either by default or by opting-in. Since `check_contact` needs to know what bodies you are interested in, you might be better off building these running tallies based on everything `get_contacts` returns? Or else all pairs of observables? I'm not quite sure the best way to build this in.

shaoyifei96, 2025-02-12:
Could you share how you checked contact? I am not able to find contact between the robot and anything like the door at all it seems

This is what I am adding in dataset_playback to check for contacts

`        # Print contacts for current state
        print(f"\nContacts at step {i+1}/{traj_len}:")
        for i, entity1 in enumerate(all_entity_names):
            for entity2 in all_entity_names[i+1:]:
                # Get the corresponding objects from env
                if entity1 in [env.robots[0].name]:
                    obj1 = env.robots[0].robot_model
                elif entity1 in list(env.fixtures.keys()):
                    obj1 = env.fixtures[entity1]
                else:
                    obj1 = env.objects[entity1]

                if entity2 in [env.robots[0].name]:
                    obj2 = env.robots[0].robot_model
                elif entity2 in list(env.fixtures.keys()):
                    obj2 = env.fixtures[entity2]
                else:
                    obj2 = env.objects[entity2]

                if env.check_contact(obj1, obj2):
                    print(f"Contact between {entity1} and {entity2}")
`

shaoyifei96, 2025-02-13:
It seems get_contact(model) works better than check_contact.

Surprisingly, when I want to check contact between cabinet and gripper, giving check_contact(cabinet, gripper) returns false since the contact geometry of the cabinet does not include the door