ARISE-Initiative/robosuite #367: No way to have initial velocities

jbonyun, 2022-12-08:
This is probably a shortcoming of Mujoco, rather than Robosuite, so feel free to close this if you don't think it is relevant.

I needed a simulation in which an object starts with an initial velocity -- a ball flying through the air. I was unable to find a way to give an object an initial velocity, in the xml or in code.

My workaround was to create an actuator that would only be active on the first timestep, and would apply the right calculated force to accelerate the object's mass to the desired velocity over the length of the timestep. Specifically we set the force in the override of `Environment.step` to set the `sim.data.ctrl` for the actuator to either `0` or `speed * mass / timestep_length`. This seems like a poor workaround.

Much better would be to have a tag in the xml/code to give velocities. This would enable simulation of more dynamic tasks.

zhuyifengzju, 2023-03-26:
Thanks for the suggestion. We will discuss this internally in the robosuite team and decide if we want to open up this option for users.

kevin-thankyou-lin, 2024-09-07:
I believe we can also directly set velocities in something like `env.sim.data.qvel[:] = xyz`, but this indeed isn't supported (currently)